MarlinFirmware/Marlin #20041: [BUG] Temperature and Extruder Commands Appear Parsed Out of Order

yysh12, 2020-11-06:
### Bug Description

I've attached sample gcode and configuration files.
Things to note: 
In this gcode file begining: hotend and bed heat up first, filament is retracted some, G28, G29, unretract, and then start printing. 
In this gcode file ending: hotend and bed turned off

Basically, when starting this print via SD card, absolutely no problems. 
When running this print via USB (octoprint):
Occasionally, the extruder will attempt to retract before hotend finishes heating up. This will cause filament not to move as it is stuck in hotend due to no heat. This will then trigger filament runout script due to filament motion sensor detecting no filament movement. Additionally, sometimes the hotend is turned off prior to the final few print moves, resulting in cold extrusion and print clogs. Basically, it appears as if some of the commands are executed out of the supplied order. The weird part is also that running the same file does not always cause same error. This error occurs sporadically. Sometimes it occurs sometimes it doesn't. Once again, all with same gcode and no changes whatsoever on my part. 
Another error, which may or may not be related, is that after stopping print manually from LCD with host actions enabled, stops the print but then continues the next few print moves with cold extrustion

My (totally uninformed) theory is that has something to possibly do with emergency parser, since that is the only out of order execution option I know of, but I have no proof of this. Is anyone else experiencing this issue? Or is this possibly user error on my part?

### Configuration Files

[Config_Gcode.zip](https://github.com/MarlinFirmware/Marlin/files/5498740/Config_Gcode.zip)

### Steps to Reproduce

1. Run supplied Gcode
2. Error may or may not occur either at beginning or ending.

#### Additional Information

For now I have switched completely to SD printing to circumvent this issue altogether. 

My noob theories as to source of error:
Emergency Parser
#define BLOCK_BUFFER_SIZE 128
#define BUFSIZE 32
Maybe the buffer sizes are too large?

ellensp, 2020-11-06:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-06 by ellensp

yysh12, 2020-11-06:
Wait I don't understand. So is this definitely something that is user error?

ellensp, 2020-11-06:
Unknown, but reads like support request.
The following line is particularly relevant 
"After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report."
In particularly in the Marlin Discord server. If the consensus is it really a bug, this can be re-opened