MarlinFirmware/Marlin #20045: [BUG] SWITCHING_EXTRUDER - disabled motor for E1

DrumClock, 2020-11-06:
**In bugfix 2.0. z DISTRIBUTION DATE "2020-11-06"**
 when using the `SWITCHING_EXTRUDER` function, 
the motor E0 STEPPER MOTOR will be disabled 

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5500131/Configuration.zip)

DrumClock, 2020-11-06:
If it disables the `#elif EXTRUDERS! = 2` condition in **SanityCheck.h**
E-STEPPER for E1 works properly.

```
#if ENABLED (SWITCHING_NOZZLE)
   #if ENABLED (DUAL_X_CARRIAGE)
     #error "SWITCHING_NOZZLE and DUAL_X_CARRIAGE are incompatible."
   #elif ENABLED (SINGLENOZZLE)
     #error "SWITCHING_NOZZLE and SINGLENOZZLE are incompatible."
   // # elif EXTRUDERS! = 2
     #error "SWITCHING_NOZZLE requires exactly 2 EXTRUDERS."
   #elif NUM_SERVOS <1
     #error "SWITCHING_NOZZLE requires NUM_SERVOS> = 1."
   #endif
```

ellensp, 2020-11-06:
These are not bugs, you trying to make marlin do things it was not designed to do. This is a feature request.

Ie marlin does not know how to have two switched 2-IN-1-OUT extruders.

DrumClock, 2020-11-06:
OK, thanks

ellensp, 2020-11-06:
You didn't have to close it... maybe someone will like this idea and implement it.   Just change Subject to FR not BUG.

DrumClock, 2020-11-06:
Hi @ellensp 
Never mind, I have FR https://github.com/MarlinFirmware/Marlin/issues/18429
and he would solve everything I have a problem with here,
Unfortunately, I don't know if anyone is working on it - that's a disadvantage to Gihub, not feedback

DrumClock, 2020-11-06:
Hi @ellensp
This is rather worse with this problem  ...... https://github.com/MarlinFirmware/Marlin/issues/19178

DrumClock, 2020-11-07:
I tried this:
in ..... \ Marlin \ src \ inc \ **SanityCheck.h** I disabled (so it can be compiled)

```
#if ENABLED (SWITCHING_NOZZLE)
  #if ENABLED (DUAL_X_CARRIAGE)
    #error "SWITCHING_NOZZLE and DUAL_X_CARRIAGE are incompatible."
  #elif ENABLED (SINGLENOZZLE)
    #error "SWITCHING_NOZZLE and SINGLENOZZLE are incompatible."
  #elif EXTRUDERS! = 2
    // # error "SWITCHING_NOZZLE requires exactly 2 EXTRUDERS."
  #elif NUM_SERVOS <1
    #error "SWITCHING_NOZZLE requires NUM_SERVOS> = 1."
  #endif

```
then I **enabled SWITCHING_NOZZLE** and set the unused **servo0**

```
#define SWITCHING_NOZZLE
#if ENABLED (SWITCHING_NOZZLE)
  #define SWITCHING_NOZZLE_SERVO_NR 0
  // # define SWITCHING_NOZZLE_E1_SERVO_NR 2 // If two servos are used, the index of the second
  #define SWITCHING_NOZZLE_SERVO_ANGLES {45, 135} // Angles for E0, E1 (single servo) or lowered / raised (dual servo)
#endif
```

**After these modifications, both motors for extruders and servos for E0-E3 switched and worked properly.**

Perhaps this will help eliminate the bug or add FW.

I am very happy to edit before PR ...... just write me what to change here.

ElectroFlux-SS, 2020-11-07:
Try disabling the following line:
#define DISABLE_INACTIVE_EXTRUDER

DrumClock, 2020-11-07:
Hi @ElectroFlux-SS
Yes this setting works.
// #define DISABLE_INACTIVE_EXTRUDER

closed 2020-11-07 by DrumClock