MarlinFirmware/Marlin #20048: [BUG] Case light blinks

G-Pereira, 2020-11-06:
### Bug Description

When I try to turn case lights on they blink and stay off.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5501874/Configuration.zip)

I am using **BTT SKR 1.3** and I am assigning a free hotend pin (P2_04)

### Steps to Reproduce

1. Send M355 S1 P255

**Expected behavior:** Light should turn on.

**Actual behavior:** Blinks very fast and stays off

#### Additional Information

No thing to add.

ellensp, 2020-11-06:
This is a configuration issue
you still have FAN1_PIN set to pin P2_04 so the instant the fan code is run it turns the pin off again.
Add #define FAN1_PIN -1 to your Configuration.h 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-06 by ellensp

G-Pereira, 2020-11-07:
Thank you anyway, it worked!