MarlinFirmware/Marlin #20061: [BUG] mini12864 with Fysetc S6 and TMC2130 very unstable

LinFor, 2020-11-07:
# Bug Description

I have mostly the same issue as #16059. Display has some glitches when in standby but after print starts (when active axis movement starts) mini12864 goes very unstable (lost contrast, lost picture, accidentally flashes and so on).
Tried descrease SPI speed (up to 2.5kHz) with no luck. Tried to tried to shield the wires - no luck.

But today I try to change SPI_MODE0 to SPI_MODE3 (in u8g_com_stm32duino_hw_spi_fn) to sync with TMC2130 mode and for the fifth time I start printing, I don't see the screen off. Some glitches like pixel shifting accidentally appears, but it's not critical. With default 2.5MHz SPI clock.

Seems that u8g_com_stm32duino_hw_spi_fn must allow override SPI_MODE and maybe speed through Configuration.h. I could have made a pull request, but the code is in another repository.

### Configuration Files

Notable: BOARD_FYSETC_S6, FYSETC_MINI_12864_2_1, TMC2130 (on axis X Y Z E0), MAX6675, all onto hw-spi.
Trying to move MAX6675 onto separate spi, but problem still exists.

Full configuration is here: https://github.com/LinFor/Marlin/tree/e-bot-s6

### Steps to Reproduce

BOARD_FYSETC_S6, FYSETC_MINI_12864_2_1, TMC2130 (on axis X Y Z E0)

**Expected behavior:**

Display working stable.

**Actual behavior:**

Display very unstable.

#### Additional Information

This can be a comment to #16059, but it's closed.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

LinFor, 2020-11-07:
With SPI_MODE3 there problem with selection of files on SD card. Screen with file list very damaged (many partial line shifts), but printing from SD card works well (I have SD_CHECK_AND_RETRY enabled).

sjasonsmith, 2020-11-07:
You might look in the SKR Pro pins file to see how that screen is configured. There were lots of quirks getting things working with various combinations of SD options. I think I got most things working, but didn’t apply them to the S6. Primarily it often has to choose to use Software SPI for things to work. 

@rhapsodyv has been working toward a more powerful SPI interface in Marlin that would likely help with this and reduce the need to fall back to SW SPI.

LinFor, 2020-11-09:
@sjasonsmith seems that pinout configuration for S6 made from SKR Pro pinout file. I don't see any differences that makes sense.
S6 hasn't onboard sd card port, so there are the only choice of SD configuration.
Now I tried Software SPI, nothing changed in a positive way.

LinFor, 2020-11-23:
I will add that the proposed solution is not complete.
I use this screen with SPI_MODE3, and it works much more stable than with SPI_MODE0, but still far from ideal. I have never seen disconnections during printing with SPI_MODE3, but sometimes the picture is scattered.
100% reproducible case of picture scattering - viewing the contents of the SD card (scrolling of the file name is turned on and there is a constant appeal to the SD card to draw the file name) - screenshot attached:
[image]
So seems that problem caused by spi mode conflict between sd card and lcd screen, or maybe incorrect reconfiguration of spi mode (cs signal control or something else).

rhapsodyv, 2020-11-23:
Yes. Not all marlin spi code was prepared to multi mode. I'm fixing it, but it will take sometime until everything is finished. I added this issue for me, and as soon as I push more SPI fix, I will tell you.

LinFor, 2020-11-23:
@rhapsodyv yes that would be great.
I am ready to help in solving as much as I can, maybe some testing or spi signals logging or anything else.

rhapsodyv, 2020-11-23:
I did a review, and a lot of place share the SPI config (mode, speed, etc). We are changing it. The problem is: that's a big change. So we are doing it parts. I will see if I can prioritise your issue.

Here is the review I did: https://github.com/rhapsodyv/Marlin/wiki/Marlin-SPI-Review

LinFor, 2020-12-24:
> This issue has had no activity in the last 30 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 7 days.

It's actual.

LinFor, 2021-01-25:
It's actual.

closed 2021-03-05 by github-actions[bot]