MarlinFirmware/Marlin #20066: [FR] custom bootscreen for new Marlin UI for TFT screens

DanJunior78, 2020-11-07:
### Description

Gain the possibility to add a second bootscreen picture  or text to add device relevant data during boot up. 
Maybe add on the main Marlin bootscreen the actual version,like for the LCD based devices.

### Feature Workflow

1. If possible keep Marlin bootscreen with adding Marlin Version to it
2. Add possibility for a custom picture which could be attached, like before (_bootscreen.h)

#### Additional Information

Same like for the LCDs 12864.
I put on this bootscreen production relevant data, like USB hub port and device name.
There would also be a text message enough. However it could be solved on a easy way.

sjasonsmith, 2020-11-07:
@DanJunior78 took the time to explain to us on Discord why this is more than just a cosmetic issue.

In a print farm environment it can be helpful to have identifying information which is easily available without extra menu navigation. A quick power on tells you what you need.

ETE-Design, 2020-11-08:
Would also be great with Custom StatusScreen, that way you can give the printers numbers / id at a print farm...

SHBnik, 2020-11-10:
as far as I know you can check the version in the "about" part, but you can make your own boot screen. For example  for mks robin nano v1.2 and v2 in the assets file replace the bmp_logo.binto your custom image.

DanJunior78, 2020-11-10:
We speak about Marlin UI, not custom MKS UI. MKS UI isn't usable,cause of main screen not showing any details about the actual print. But you're right.