MarlinFirmware/Marlin #20067: G26 Validate mesh missing in LCD menu

tarsonis123, 2020-11-07:
So I was happy to find that feature but I can't find it anywhere in the menu after mesh leveling. I'm using the latest Marlin-bugfix-2.0.x repository files. I was researching if there are any specific condition to show it in the menu but didn't find anything. Though I can run the command through Pronterface so its activated. Sorry if its intended not be seen in the menu and I just didn't get the memo.

boelle, 2020-11-07:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

tarsonis123, 2020-11-07:
[config_Mesh Leveling Enabled.zip](https://github.com/MarlinFirmware/Marlin/files/5505257/config_Mesh.Leveling.Enabled.zip)

Oh I didn't see that template request. Sorry about that. The Marlin repo I use is from that link and I got the state from yesterday. I'm gone test it after print is done.

thisiskeithb, 2020-11-07:
`G26_MESH_VALIDATION` simply enables the `G26` G-code.

If `G26_MESH_VALIDATION` and `AUTO_BED_LEVELING_UBL` are enabled, an extra UBL tools menu with a "Validate Mesh" option is added, but you have `MESH_BED_LEVELING` enabled in your config.

Switch to `AUTO_BED_LEVELING_UBL` or enable/use `CUSTOM_USER_MENUS` to add this functionality to your build.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-07 by thisiskeithb