MarlinFirmware/Marlin #20068: Creality CR6SE Strain Gauge status

solvan230, 2020-11-07:
I've searched everywhere but couldn't find any single post on how to get Strain Gauge status information.
What i'd like to do is if Strain Gauge doesn't get any signal/input for more than 20 seconds after laying first layer, it would send X,Y, Z to a specific location or pause the print. It would be awesome if someone could shed some light. Thank you

sjasonsmith, 2020-11-07:
I think that just behaves like a BLTouch, you can’t just read it like a normal strain gauge.

I don’t understand what you expect to detect? Do you expect to detect the nozzle bumping the print?

solvan230, 2020-11-07:
Yes, that’s right. I was trying to prevent the nozzle bumping the print. But couldn’t find any gcode command “if condition” for it. 
All I could find is M42 Pin 5 might has something to do with it. But no idea how to make it work ??

If it behaves as BLtouch is there anyway to add GCode command with “if condition” ?

sjasonsmith, 2020-11-07:
I believe @thinkyhead has done the work for that sensor, although I am not certain.

I think I heard something about needing to ignore endstops except during the actual probe. I suspect it may generate positives if you tried to use it all the time.

I don’t believe your request will be practical with this hardware, but will wait for someone with direct experience to respond.

ChipCE, 2020-11-09:
There is no Pause support atm but <code>SD_ABORT_ON_ENDSTOP_HIT</code> can be use to stop the print.

solvan230, 2020-11-10:
The purpose is to have it pause automatically so it would act as a detector.

The led light, light up when I push the hotend. Maybe that could be used as well but how to interpret that input? I couldn’t see it on octoprint terminal ?

Sebazzz, 2020-11-24:
Pull request #19958 I've submitted is integrating changes from CR-6 community firmware back into Marlin

> What i'd like to do is if Strain Gauge doesn't get any signal/input for more than 20 seconds after laying first layer, it would send X,Y, Z to a specific location or pause the print. It would be awesome if someone could shed some light. Thank you

I think you mean to use this as some kind of failure detection? I think the strain gauge system is too inaccurate to to that. Its function very much relies on how the firmware tares the probe in combination with the specific setting on the little potmeter inside the hot-end. I don't think it is feasible to use it for anything.

dryphi, 2020-11-24:
What about using the StallGuard function on the TMC2209 drivers for this?

Sebazzz, 2020-11-24:
StallGuard is not suitable for this I think:

> StallGuard4 does not operate reliably at extreme motor velocities: Very low motor velocities (for many
motors, less than one revolution per second) generate a low back EMF and make the measurement
unstable and dependent on environment conditions (temperature, etc.)

dryphi, 2020-11-24:
Might be worth a try:
https://marlinfw.org/docs/gcode/M914.html
With using multiple drivers, (X, Y, Z, E), there's a greater chance one of them might trigger, if it halts when it's not supposed to.
Just a thought.

dryphi, 2020-11-24:
It works just like an endstop. It's binary, just two wires.
https://www.youtube.com/watch?v=hs6IVfNrf5k

sjasonsmith, 2020-11-24:
For what it's worth, there are changes for this printer (including STRAIN_GAUGE_PROBE) in this PR. I don't know if it is relevant to this discussion or not.
https://github.com/MarlinFirmware/Marlin/pull/19958

Sebazzz, 2020-11-24:
> The CR-6 SE implementation uses FIX_MOUNTED_PROBE (so yes it does work like a BLTouch)
> The intended result is the same - activation of the Z_MIN_PIN which is PA4 if I understand correctly.

It is just a simple signal that is pulled high or low. But the strain gauge is a bit more intelligent though, because it can be tared from firmware to compensate for the bowden tube pulling on the hot-end.

solvan230, 2020-12-01:
Thanks for the suggestions. Tried the Stallguard method but unfortunately it’s not accurate enough to stall if it hits the print. 
Interestingly upon trying different scenarios, I got an idea if only I could tap the strain gauge sensor input output to cancel the print if it’s printing on midair(to prevent printing spaghetti if you wanna call it). 
I notice when the printer is printing, the strain gauge led light switching on and off because is slightly touching the print. But completely off when printing spaghetti or printing on the air. So I guess this strain gauge has potentials not only prevent bumping the print but also prevent spaghetti!!!
Can’t waitttttt to make it work so I could rest comfortably. Lol

Sebazzz, 2020-12-01:
> I got an idea if only I could tap the strain gauge sensor input output to cancel the print if it’s printing on midair(to prevent printing spaghetti if you wanna call it).

I have been observing my strain gauge for the past couple of prints. When printing, it always lights up a little. That can be because it hits the printed piece, but more likely the bowden tube is pulling on the print head.

thisiskeithb, 2022-04-09:
This will be added in https://github.com/MarlinFirmware/Marlin/pull/19958

closed 2022-04-09 by thisiskeithb