MarlinFirmware/Marlin #20070: [BUG] UBL with GRID_MAX_POINTS_X=15: saving mesh causes reboot

salfter, 2020-11-08:
### Bug Description

I've been getting bed leveling dialed in on my Hypercube 300.  I've had UBL enabled since the start, with GRID_MAX_POINTS_X set to 10.  When I increased it to 15, G29 S1 appears to start saving the mesh, but then Marlin crashes and the board reboots.  I backed off GRID_MAX_POINTS_X to 14 and tried G29 S1 before probing the bed, just to verify that it would work...it did.  Is there some statically-allocated resource that's getting stomped?

### Configuration Files

[marlin-config.zip](https://github.com/MarlinFirmware/Marlin/files/5505797/marlin-config.zip)

### Steps to Reproduce

1. Enable AUTO_BED_LEVELING_UBL and set GRID_MAX_POINTS_X to 15.  Build and install firmware
2. Send G29 S1 to the printer.  (It's not necessary to do G29 P1 first; the bug is triggered either way.)
3. Observe that it reboots.

**Expected behavior:**

The bed probing results get stored to whatever nonvolatile storage your board has.

**Actual behavior:**

Marlin crashes and restarts.

salfter, 2020-11-08:
More testing results:

GRID_MAX_POINTS_X=14

Empty grid: saves OK.
Grid obtained by G29 P1: saves OK.
Fill out the empty spots in the grid with two iterations of G29 P3, then try saving: crashes.

ellensp, 2020-11-08:
as a test disable #define USE_WATCHDOG, see it it still reboots
Suspect UBL is taking to long triggering watchdog timer..

salfter, 2020-11-08:
> as a test disable #define USE_WATCHDOG, see it it still reboots

No reboot.  With GRID_MAX_POINTS_X=15, after probing the bed and filling in the empty spots, saving the mesh took about 5.5 seconds.

ellensp, 2020-11-08:
defaults is 4 second timeout, in bugfix WATCHDOG_DURATION_8S  was added 3 days ago https://github.com/MarlinFirmware/Marlin/commit/4fe1adc383283b10a4baa60ad4e4d3306b0957c6

thinkyhead, 2020-11-08:
Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

salfter, 2020-11-09:
> Please test the [`bugfix-2.0.x` branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands.

I tried, but now I'm getting "TMC CONNECTION ERROR"s on startup (I use TMC2130s with SPI configuration).  If I revert to the release branch, it works properly.  I've reconstructed settings since the config file version changed; the configs I'm using with the bugfix branch are attached: [marlin-config-bugfix.zip](https://github.com/MarlinFirmware/Marlin/files/5507412/marlin-config-bugfix.zip)

salfter, 2020-11-09:
> TMC CONNECTION ERROR

...and that may have been the fault of not having TMC_USE_SW_SPI defined.  Trying again...

salfter, 2020-11-09:
The bugfix branch, with WATCHDOG_DURATION_8S defined in Configuration_adv.h, gets the job done...guess I'll be switching to it for a while.

closed 2020-11-09 by salfter

casterle, 2020-11-10:
I had the same issue; the bugfix branch was the solution.