MarlinFirmware/Marlin #20082: [BUG] During leveling, BLTouch trigger is seemingly detected properly, but z move does not stop, resulting in failure

cohaolain, 2020-11-10:
### Bug Description

Hi, I recently installed a genuine BLTouch v3.1 on my Ender 3 v2.
I have intermittently experienced, during a downward z move during G29 levelling, the probe will trigger, but the z-move downwards will fail to stop.

I've verified the issue isn't an issue with the probe (oscilloscope shows triggering pulse when the nozzle touches the bed, when this failure does occur).
This also doesn't seem to be an issue with my physical wiring etc., since 
1. **it works most of the time**
2. I've used PINS_DEBUGGING and enabled leveling debugging output to try and find a fix for this myself, and seen that a **"Z_MIN_PROBE:1" is logged when the probe touches the bed as expected**. It's just that this never seems to be acted on (at least, it never stops ploughing the z-axis down further into the bed...)

### Configuration Files

Attached here: [Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5514092/Configuration.zip)

No modifications, except adding more debugging output for the purposes of trying to address the issue myself.

### Steps to Reproduce

1. Home with G28
2. Run ABL with G29. Personally, I use a 3x3 grid with double probing.
  i. Using high-speed mode flag (have found changing this makes no impact)

**Expected behavior:**

Z-axis movement shoudl stop at each probe point, when the probe triggers during the downward z-move.

**Actual behavior:**

Sporadically, the z-axis will continue moving down, despite the probe being triggered and this being detected by the controller (since "Z_MIN_PROBE:1" is appearing in the logs).

#### Additional Information

I'm attaching a [serial.log](https://github.com/MarlinFirmware/Marlin/files/5514085/serial.log)
 taken from octoprint. It is quite verbose, as it has pin debugging and leveling debugging enabled.
**Serial.log**: [GitHub Attachment](https://github.com/MarlinFirmware/Marlin/files/5514085/serial.log)
I'm also attaching a [video](https://youtu.be/RoC2-VB2-4A?t=251) to show what is actually going on here. In this case, the first point fails, though different points fail different times.
**Video:** https://youtu.be/RoC2-VB2-4A?t=251

ellensp, 2020-11-10:
In Configuration_adv.h  Enable this
 // Feature: Switch into SW mode after a deploy. It makes the output pulse longer. Can be useful
 //          in special cases, like noisy or filtered input configurations.
 //#define BLTOUCH_FORCE_SW_MODE

shitcreek, 2020-11-10:
From the video, it doesn't look like your probe is working properly. It should automatically retracts upon touching the bed. Have you done an `M401` and try trigger the probe by touching it and seeing if it behaves as it should?

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

cohaolain, 2020-11-10:
> In Configuration_adv.h Try enabling this
> // Feature: Switch into SW mode after a deploy. It makes the output pulse longer. Can be useful
> // in special cases, like noisy or filtered input configurations.
> //#define BLTOUCH_FORCE_SW_MODE

Hey, @ellensp
Thanks, I'll try this in the morning! Though IIRC, my oscilloscope already showed a longer pulse (possibly because I'm in high speed mode and the probe stays down between probes because of that - to clarify, use of high speed mode doesn't seem to have any influence)

cohaolain, 2020-11-10:
> From the video, it doesn't look like your probe is working properly. It should automatically retracts upon touching the bed. Have you done an `M401` and try trigger the probe by touching it and seeing if it behaves as it should? 

Hey @shitcreek ,
Thanks for your reply. That's expected behaviour with the BLTouch high speed mode enabled, I believe - it removes the need to stow the probe between multiple probes and this doesn't seem to affect the issue I'm describing here (I've tested with the mode on and off).
To clarify, when I remove the high speed mode flag, the automatic stow on touching the bed **does** occur as you mentioned, but the mode was enabled in the attached video.
```
/**
   * Use "HIGH SPEED" mode for probing.
   * Danger: Disable if your probe sometimes fails. Only suitable for stable well-adjusted systems.
   * This feature was designed for Delta's with very fast Z moves however higher speed cartesians may function
   * If the machine cannot raise the probe fast enough after a trigger, it may enter a fault state.
   */
  #define BLTOUCH_HS_MODE
```

Also, I'm aware this says to disable in case of failures - but this issue isn't caused by the probe going into error state or the probe failing to trigger (since Marlin seems to see the probe getting triggered in the logs, it's just seemingly not acting upon it?). I've tried disabling, and it does not seem to stop this from happening.

shitcreek, 2020-11-10:
I see. I stand corrected. I am not familiar with that feature.
Let us know how it goes after applying Ellen's suggestion.

cohaolain, 2020-11-11:
Unfortunately, after some painstaking hours of trying to debug this myself, I've been having a pretty hard time actually reproducing this consistently (seemingly since I added a bunch more debugging messages.... possibly a concidence). It has happened again once or twice, with the same symtoms (logs show `z_min_probe:1`, i.e. detects the trigger, but z keeps moving down).
I've yet to try `#define BLTOUCH_FORCE_SW_MODE`, I do hope to get around to this. Though I have a hunch that while a longer pulse might be better _detected_... the case remains that the logs show the probe trigger *is* currently being detected, seemingly before we start moving the z-axis back up.
I added some debug modifications to log what the z-axis location is at  the moment we log `z_min_probe:1` so we can correlate if we are indeed moving down _after_ the trigger point. Though down on my luck, I have been having trouble reproducing with these extra debug statements included.... it just seems to work... with these.... agh!

Interestingly I did have an issue recently where modifying my babysteps/z-offset downwards by the minimum amount during printingt would sometimes cause the z axis to move down excessively (i.e. I'd make the minimum step, and the nozzle would dive into the bed). I hadn't thought these would be related, unless perhaps there's some underlying issue affecting these movements....

Still plenty to be figured out here, it seems :weary:

cohaolain, 2020-11-15:
Hey @ellensp @shitcreek , sorry for taking so long to get back (been a long week, and weekend as it turned out).

I was able to test ellensp's suggestion, adding `BLTOUCH_FORCE_SW_MODE`, unfortunately this hasn't changed my results at all (failed probes appear to pulse correctly, for quite some time until the z-probe).
I've also disabled `BLTOUCH_HS_MODE`, just to remove the extra variable, also seeing no change. My probe is retracting on successful probes as expected.

Adding my own debugging output seems to make the issue particularly hard to reproduce, as does, it seems, enabling debugging in general. At least compared to how often the issues seems to occur for me with all debugging disabled, as in stock.

See some new logging output here:
([__corresponding video, with oscilloscope showing the pulses and then long pulse not acted upon on final failed probe__](https://photos.app.goo.gl/Ai4yX5xoGtoZZP2j6)):
```log
2020-11-15 17:51:46,107 - Recv: deploy: 1
2020-11-15 17:51:46,112 - Recv: >>> Probe::run_z_probe  X137.00 Y114.00 Z8.65
2020-11-15 17:51:46,114 - Recv: >>> Probe::probe_down_to_z  X137.00 Y114.00 Z8.65
2020-11-15 17:51:46,118 - Recv: BLTouch DEPLOY requested
2020-11-15 17:51:46,121 - Recv: BLTouch Command :10
2020-11-15 17:51:46,915 - Recv: BLTouch Command :60
2020-11-15 17:51:47,466 - Recv: bltouch.deploy_proc() end
2020-11-15 17:51:47,568 - Recv: >>> do_blocking_move_to  X137.00 Y114.00 Z8.65
2020-11-15 17:51:47,572 - Recv: >  X137.00 Y114.00 Z1.61
2020-11-15 17:51:47,966 - Recv: echo:busy: processing
2020-11-15 17:51:47,972 - Recv:  T:170.00 /170.00 B:60.00 /60.00 @:65 B@:35
2020-11-15 17:51:48,775 - Recv:   Z_MIN_PROBE:1
2020-11-15 17:51:48,777 - Recv: 
2020-11-15 17:51:49,199 - Recv: <<< do_blocking_move_to  X137.00 Y114.00 Z1.61
2020-11-15 17:51:49,204 - Recv:   current_position= X137.00 Y114.00 Z1.61 : sync_plan_position
2020-11-15 17:51:49,207 - Recv: <<< Probe::probe_down_to_z  X137.00 Y114.00 Z1.61
2020-11-15 17:51:49,210 - Recv: FAST Probe fail! - No trigger.
```
Given that

1. I can see that `Z_MIN_PROBE:1` does indeed appear in the logs
and
2. I can see that it is *after* that reading, that we are notified of the `Probe fail! - No trigger`.

My initial hunch would be that there's some form of race condition scenario, or interrupt priority shenanigans going here, maybe confined to my board type :confused:. Though I'm honestly very new to the Marlin codebase so this is a wild guess at best.

Of course, I can't rule out a hardware issue here, just yet, I suppose. I ordered another BLTouch (this time from [Amazon directly](https://www.amazon.co.uk/antclabs-bltouch-leveling-premium-printer/dp/B01FFV2TOS/), whereas [my current one](https://www.amazon.co.uk/gp/product/B087PPFNNP/) is from a marketplace seller who also supplied some extension cables, though I believe it's genuine. Unfortunately it only occurred to me after ordering that second BLTouch that I could have spent a little more and got an [SKR Mini E3 bundled with a BLTouch](https://www.amazon.co.uk/gp/product/B07Z5RHZ4Z) to be a little more thorough in my testing, though that seems a little overkill and, given the weirdness I've sifted through so far, this _seems_, so far at least, to be a Marlin issue. 

By all means, please correct me if I'm completely going in the wrong direction here, or if you have any more suggestions.

Thanks :relaxed:

cohaolain, 2020-11-16:
I have noted [a comment](https://github.com/MarlinFirmware/Marlin/pull/19301/files#r484584508) on #19301 which may be relevant.

Previously to that, I note that #18637 fixed a previous issue that appears to have exhibited very similar behaviour to what I'm now experiencing.

I may also try a slightly older version of Marlin to see if, perhaps, the issue was present after both of those^ changes but before the current bugfix branch head. Hopefully won't take too long to reproduce if the issue does indeed exist back then.

cohaolain, 2020-11-16:
Most likely have confirmed this isn't an issue with the probe itself. Have tried with a second genuine BL Touch from Amazon directly. After running quite a few more levelling, I did see the same result occur several times.

JackNewman12, 2020-11-17:
I have seen this too. 
bltouch (clone) triggers but the axis continues to move down.  Connections seem fine using a multimeter but I have yet to check the pulse using an oscilloscope. 

Interestingly I have done the probe test `M48 P50 V4` over 5 times in a row without any failures, but every second 5x5 bed level fails. 

Ender 3 V2 4.2.2 board.
Running a build from 5th Nov bugfix

cohaolain, 2020-11-23:
I have a little more logging of this issue captured since. Granted, it's more of the same, but will try upload soon;

I can see the endstop watcher (courtesy of `PINS_DEBUGGING`) logging that the probe has been triggered when it manually checks the pins, but then the z-axis continuing downwards, with the move block not being removed until about a whole second later (the `quickstop` in `stepper` never gets called when this occurs).

It's almost as if the interrupt isn't triggering, or something's going wrong during the interrupt call.

The most aggravating thing is that, if I add a debugging line to the **beginning** of `Endstops::update()`, the issue _seems_ to completely go away! I'd done this a few times to try and tell if the interrupt was getting triggered at all, but can't easily reproduce. However, with that debugging line removed (unmodified), the issue starts happening enough that I can reproduce within a few tries!

This one's really got me beat... I got the probe to save me time levelling the bed, yet I've ended up not printing anything in a little over 2 weeks as this has been bugging me so much! :cry: 

If anybody else has a better understanding of the architecture, and any potential causes, or any ideas at all, it'd be much appreciated! :smile:

cohaolain, 2020-11-23:
@JackNewman12 While the probe is moving down, the probe should turn red briefly, and then either retract immediately (so it would stay red then) or, if `BLTOUCH_HS_MODE` is enabled, the probe will move away from the bed and the light should go off again.

You can compile with the `PINS_DEBUGGING` option enabled, and then send `M43 E1` over serial. Then when your probe triggers you can see if the pulses are being detected. Though I'd imagine it is generating the pulses okay, since your probe is working most of the time (unless it's noise/loose cabling etc., i.e. a hardware issue).

cohaolain, 2020-11-24:
Hi all, 

Quick note: with a successful probe, we see **"Completed quick stop"** called after endstop interrupt triggers this action in `stepper.cpp`.
Then,**"Blocking move ended"** refers to one of
1. probe stopped by quick stop in interrupt (desired)
2. probe got to minimal z position but was never stopped by an interrupt from the probe (failure - **"no trigger"**)
always called after `do_blocking_move_to_z(z, fr_mm_s)` in `probe.cpp`.

The issue is that the failure case is occuring (but not for the lack of a trigger) - `Z_MIN_PROBE` does indeed read as triggered, according to endstop watcher, well in advance of the move ending stopping!
i.e. **Can see the state of `Z_MIN_PROBE` changing using endstop watcher, but the endstop interrupt is not stopping the stepper.**
### First example
```
2020-11-22 21:53:37,156 - Recv:   Z_MIN_PROBE:0
2020-11-22 21:53:37,156 - Recv: 
2020-11-22 21:53:37,219 - Recv: echo:busy: processing
2020-11-22 21:53:37,224 - Recv:  T:166.11 /215.00 B:51.80 /60.00 @:127 B@:127
2020-11-22 21:53:39,223 - Recv:  T:168.30 /215.00 B:52.17 /60.00 @:127 B@:127
2020-11-22 21:53:39,225 - Recv: echo:busy: processing
# In success case, interrupt sees endstop trigger first, stops stepper, move is ended, all before endstop watcher sees the trigger.
2020-11-22 21:53:39,605 - Recv: Completed quick stop in stepper
2020-11-22 21:53:39,607 - Recv: Blocking move ended 
2020-11-22 21:53:39,662 - Recv:   Z_MIN_PROBE:1
2020-11-22 21:53:39,662 - Recv: 
2020-11-22 21:53:39,919 - Recv:   Z_MIN_PROBE:0
2020-11-22 21:53:39,919 - Recv: 
# In fail case, endstop watcher sees the probe trigger here, but interrupt never carried out.
2020-11-22 21:53:41,076 - Recv:   Z_MIN_PROBE:1
2020-11-22 21:53:41,077 - Recv: 
2020-11-22 21:53:41,223 - Recv:  T:170.62 /215.00 B:52.54 /60.00 @:127 B@:127
2020-11-22 21:53:41,225 - Recv: echo:busy: processing
2020-11-22 21:53:42,853 - Recv: Blocking move ended 
2020-11-22 21:53:42,855 - Recv: SLOW Probe fail! - No trigger.
2020-11-22 21:53:43,223 - Recv:  T:172.77 /215.00 B:52.94 /60.00 @:127 B@:127
2020-11-22 21:53:43,225 - Recv: echo:busy: processing
2020-11-22 21:53:45,223 - Recv:  T:174.69 /215.00 B:53.35 /60.00 @:127 B@:127
2020-11-22 21:53:45,226 - Recv: echo:busy: processing
2020-11-22 21:53:47,632 - Recv:   Z_MIN_PROBE:0
2020-11-22 21:53:47,632 - Recv: 
```
### Another example
Here, similarily, the probe is triggered (seen by endstop watcher), well before move ends - but never acted upon by stopping the steppers.
```
2020-11-22 22:54:45,541 - Recv:   Z_MIN_PROBE:0
2020-11-22 22:54:45,541 - Recv: 
2020-11-22 22:54:46,827 - Recv:   Z_MIN_PROBE:1
2020-11-22 22:54:46,827 - Recv: 
2020-11-22 22:54:47,071 - Recv: echo:busy: processing
2020-11-22 22:54:47,425 - Recv:  T:215.16 /215.00 B:59.97 /60.00 @:84 B@:38
2020-11-22 22:54:48,512 - Recv: Blocking move ended 
2020-11-22 22:54:48,514 - Recv: SLOW Probe fail! - No trigger.
2020-11-22 22:54:49,071 - Recv: echo:busy: processing
2020-11-22 22:54:49,425 - Recv:  T:215.10 /215.00 B:59.99 /60.00 @:85 B@:34
2020-11-22 22:54:51,071 - Recv: echo:busy: processing
2020-11-22 22:54:51,425 - Recv:  T:215.00 /215.00 B:60.00 /60.00 @:87 B@:31
2020-11-22 22:54:53,382 - Recv:   Z_MIN_PROBE:0
2020-11-22 22:54:53,382 - Recv: 
```

rhapsodyv, 2020-12-04:
Anyone having this issue can reach me at discord?

Oninaig, 2020-12-15:
Want to chime in and say I am having the same issue on an ender 3 v2 with a bltouch 3.1. I've swapped the bltouch already to rule out the bltouch as the issue but my symptoms are the same. I can do a probe deviation test 50 times without issue but trying to complete a full G29 P1 results in a failure along the way where the probe will trigger but the z axis will keep moving down. 

With debug leveling for the bed, I get the following output:

```
Recv: <<< do_blocking_move_to  X159.22 Y83.89 Z9.32
Recv: >>> Probe::probe_down_to_z  X159.22 Y83.89 Z9.32
Recv: BLTouch DEPLOY requested
Recv: BLTouch Command :10
Recv: BLTouch Command :60
Recv: bltouch.deploy_proc() end
Recv: >>> do_blocking_move_to  X159.22 Y83.89 Z9.32
Recv: >  X159.22 Y83.89 Z-10.70
[...]
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Operational" to "Offline (Error: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3831)"
Connection closed, closing down monitor
```

I am going to enable `PINS_DEBUGGING` to get a better log of the actual pin being triggered. Standby

SondreNjaastad, 2021-01-15:
Hi, i'm having this issue with a Ender 3 v2 and original brand new BLTouch. I have yet just experienced it from OctoPrint and not yet from Cura on Windows directly connected to the printer over USB. Did anyone manage to resolve it?

closed 2021-02-23 by github-actions[bot]