MarlinFirmware/Marlin #20083: [BUG] (short description)

soukrame, 2020-11-10:
### Bug Description

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:**

**Actual behavior:**

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

closed 2020-11-10 by thisiskeithb

soukrame, 2020-11-10:
[]()

ellensp, 2020-11-10:
Empty issue. Didn't follow template so was closed.