MarlinFirmware/Marlin #20085: [BUG] (short description)

Onur-Altuntas, 2020-11-10:
### Bug Description

When I uncomment **Power Loss Recovery** in Configuration_adv.h file, I get compling error about FANS_LOOP(i). 

I tried both Arduino IDE and PlatfromIO. Result is the same. I was getting the same error with 2.0.54. I thought that it was solved in 2.0.72 but I still have the same problem.

### Configuration Files

[Marlin Configurations.zip](https://github.com/MarlinFirmware/Marlin/files/5515073/Marlin.Configurations.zip)

My actual configuration is:

  #define POWER_LOSS_RECOVERY
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_ENABLED_DEFAULT   true // Power Loss Recovery enabled by default. (Set with 'M413 Sn' & M500)
    //#define BACKUP_POWER_SUPPLY       // Backup power / UPS to move the steppers on power loss
    //#define POWER_LOSS_RECOVER_ZHOME  // Z homing is needed for proper recovery. 99.9% of the time this should be disabled!
    //#define POWER_LOSS_ZRAISE       2 // (mm) Z axis raise on resume (on power loss with UPS)
    //#define POWER_LOSS_PIN         44 // Pin to detect power loss. Set to -1 to disable default pin on boards without module.
    //#define POWER_LOSS_STATE     HIGH // State of pin indicating power loss
    //#define POWER_LOSS_PULL           // Set pullup / pulldown as appropriate
    //#define POWER_LOSS_PURGE_LEN   20 // (mm) Length of filament to purge on resume
    //#define POWER_LOSS_RETRACT_LEN 10 // (mm) Length of filament to retract on fail. Requires backup power.

    // Without a POWER_LOSS_PIN the following option helps reduce wear on the SD card,
    // especially with "vase mode" printing. Set too high and vases cannot be continued.
    #define POWER_LOSS_MIN_Z_CHANGE 0.05 // (mm) Minimum Z change before saving power-loss data
  #endif

### Steps to Reproduce

1. [First Step] Uncomment Power Loss Recovery in Configuration_adv.h
2. [Second Step] Compile

**Expected behavior:**

Compiling without any error.

**Actual behavior:**

PlatfromIO terminal Output is below:

Marlin\src\feature\powerloss.cpp: In static member function 'static void PrintJobRecovery::resume()':
Marlin\src\feature\powerloss.cpp:436:13: error: 'i' was not declared in this scope
   FANS_LOOP(i) {
             ^
Marlin\src\feature\powerloss.cpp:436:14: error: 'FANS_LOOP' was not declared in this scope
   FANS_LOOP(i) {
              ^
At global scope:
cc1plus.exe: warning: unrecognized command line option '-Wno-register'
*** [.pio\build\mega2560\src\src\feature\powerloss.cpp.o] Error 1

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

sjasonsmith, 2020-11-10:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

ellensp, 2020-11-10:
You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN. This is impossible you cant have software control over the fan and have it auto turn on when temp > 50c

ellensp, 2020-11-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

sjasonsmith, 2020-11-10:
@Onur-Altuntas after I requested you test with bugfix, both @ellensp and I did just that. After fixing the conflict they explained above, it builds just fine.

Onur-Altuntas, 2020-11-10:
> You cannot set E0_AUTO_FAN_PIN equal to FAN_PIN. This is impossible you cant have software control over the fan and have it auto turn on when temp > 50c

E0_AUTO_FAN_PIN is not equal to FAN_PIN. I changed the FAN_PIN as -1 in pins_RAMPS.h
I don't get any error about it during compiling.

#ifndef FAN_PIN
  #if EITHER(IS_RAMPS_EFB, IS_RAMPS_EFF)          // Hotend, Fan, Bed or Hotend, Fan, Fan
    #define FAN_PIN                 -1 // RAMPS_D9_PIN
  #elif EITHER(IS_RAMPS_EEF, IS_RAMPS_SF)         // Hotend, Hotend, Fan or Spindle, Fan
    #define FAN_PIN                 RAMPS_D8_PIN
  #elif ENABLED(IS_RAMPS_EEB)                     // Hotend, Hotend, Bed
    #define FAN_PIN                            4  // IO pin. Buffer needed
  #else                                           // Non-specific are "EFB" (i.e., "EFBF" or "EFBE")
    #define FAN_PIN                 RAMPS_D9_PIN
  #endif
#endif

I just tested Marlin-bugfix-2.0.x. Still the same.

sjasonsmith, 2020-11-10:
Attach you pins file and any other file you’ve modified, and we can see why it isn’t compiling.

Onur-Altuntas, 2020-11-10:
Please find the pins file below.
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/5518696/pins_RAMPS.zip)

ellensp, 2020-11-11:
With the additional information of setting FAN_PIN -1  I can replicate this on current bugfix.
Can be replicated from stock bugfix with these changes
enable POWER_LOSS_RECOVERY
enable SDSUPPORT  needed by  POWER_LOSS_RECOVERY
define FAN_PIN -1

ellensp, 2020-11-11:
PR to fix this created.

Onur-Altuntas, 2020-11-11:
> PR to fix this created.

Thank you for quick response and solution.

_"Added a #if HAS_FAN block around offending code."_

It's now compiled with above motification. Perfect!

closed 2020-11-11 by thinkyhead