MarlinFirmware/Marlin #20090: [BUG] TFT_LVGL_UI compile fails with Robin Nano v1.2, Marlin bugfix-2.0.x

BeansIsFat, 2020-11-10:
### Bug Description

Compiling the bugfix-2.0.x branch with TFT_LVGL_UI enabled fails with the following error:

Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp:71:25: error: 'suicide' was not declared in this scope

### Configuration Files

ZIP file containing `Configuration.h` and `Configuration_adv.h` is attached.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5518522/config.zip)

### Steps to Reproduce

1. Checkout bugfix-2.0.x branch
2. Enable TFT_LVGL_UI
3. Set default_envs to mks_robin_nano35 in platform.ini
4. Compile

**Expected behavior:**

Successful compile

**Actual behavior:**

Failed compile with the following error message:  
Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp:71:25: error: 'suicide' was not declared in this scope

ellensp, 2020-11-11:
Cannot replicate this issue, either with provided configs or by following Steps to reproduce.
Try new bugfix, as it does change hourly.

thisiskeithb, 2020-11-11:
> Cannot replicate this issue, either with provided configs or by following Steps to reproduce.
> Try new bugfix, as it does change hourly.

Agreed. Closing.

---

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

thisiskeithb, 2020-11-11:
Reopening after further discussion on Discord. I've asked OP to upload *all* their modified files since they had also modified the pins file to include TMC support and `MKSPWC`.

BeansIsFat, 2020-11-11:
Here is the pins_MKS_ROBIN_NANO.h file (and the platformio.ini file for good measure). These two files, and the config files above, are the only modified files in my environment.
[pins_MKS_ROBIN_NANO.h and platformio.ini.zip](https://github.com/MarlinFirmware/Marlin/files/5526069/pins_MKS_ROBIN_NANO.h.and.platformio.ini.zip)

thisiskeithb, 2020-11-11:
<details><summary>Full error output with MKSPWC enabled:</summary>

```prolog
In file included from Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/boards.h:24:0,
                 from Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/MarlinConfigPre.h:37,
                 from Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp:22:
Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp: In function 'void event_handler(lv_obj_t*, lv_event_t)':
Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp:71:25: error: 'suicide' was not declared in this scope
       TERN(HAS_SUICIDE, suicide(), queue.enqueue_now_P(PSTR("M84")));
                         ^
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:419:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:196:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/wifi_module.cpp.o
                             ^~~~~~~
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:195:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:191:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
                             ^~~~~
Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp:71:7: note: in expansion of macro 'TERN'
       TERN(HAS_SUICIDE, suicide(), queue.enqueue_now_P(PSTR("M84")));
       ^~~~
Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp:71:25: note: suggested alternative: 'spi_mode'
       TERN(HAS_SUICIDE, suicide(), queue.enqueue_now_P(PSTR("M84")));
                         ^
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:419:26: note: in definition of macro 'THIRD'
 #define THIRD(a,b,c,...) c
                          ^
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:196:29: note: in expansion of macro '___TERN'
 #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
                             ^~~~~~~
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:195:29: note: in expansion of macro '__TERN'
 #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
                             ^~~~~~
Marlin/src/lcd/extui/lib/mks_ui/../../../../inc/../core/macros.h:191:29: note: in expansion of macro '_TERN'
 #define TERN(O,A,B)         _TERN(_ENA_1(O),B,A)    // OPTION converted to '0' or '1'
                             ^~~~~
Marlin/src/lcd/extui/lib/mks_ui/draw_set.cpp:71:7: note: in expansion of macro 'TERN'
       TERN(HAS_SUICIDE, suicide(), queue.enqueue_now_P(PSTR("M84")));
       ^~~~
*** [.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_set.cpp.o] Error 1
```
</details>

rhapsodyv, 2020-11-20:
can you test this PR? #20214

Thanks.

BeansIsFat, 2020-11-20:
I was able to compile and use the power off feature from the GUI and with M81. It appears to be functioning properly. Thanks.

rhapsodyv, 2020-11-20:
Merged. I'm closing this. Thanks!

PR #20214 (2020-11-20): Missing header for suicide() on LVGL UI
### Description

Fix missing header to use suicide() on LVGL_UI.

### Benefits

Fix: #20090
closed 2020-11-20 by rhapsodyv