MarlinFirmware/Marlin #20094: MKS Gen L v1.0 + TMC2209 V3 UART problem

BadPanda357, 2020-11-10:
Hi all i have problem and i dont know if it is software related or hardware... I hope you can help me out.
Recently i bought MKS Gen L v1.0 board with 5 TMC2209 drivers for a cheap. Problem is i cant get them work vith UART.
The drivers are a little strange, on all TMC2209 that i found on tutorials there are 3 pins, but on mine only 2.
Drivers are fysetc 2209 v3. Here is a link https://wiki.fysetc.com/Silent2209/
As a hobbyist that schematics wont really help me coz i dont understand it...
I tried all i could, solder 1k resistor on one pin, tried with two separate wires one on tx one on rx pin, tried to swap plases, tried to remap dedicated pins, #define SOFTWARE_DRIVER_ENABLE on and off, ... no luck
Here is my configuration and some pictures. Pls help i really ned printer to work thanks in advance
[image]
[image]
[image]

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5520062/Configuration.zip)

thisiskeithb, 2020-11-10:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-10 by thisiskeithb

BadPanda357, 2020-11-13:
Hi this is not user support question... I check out and all connection is fine i connect TX and RX pins of driver to dedicated pins on board, and when i connect to board with pronterface it shows All Low... I believe this is bug or firmware issue thats why i post it here.