MarlinFirmware/Marlin #20097: [BUG] (pins_CREALITY_V427.h missing def for #if ENABLED(LCD_FOR_MELZI))

BigBearBrian, 2020-11-11:
### Bug Description

Missing definitions for LCD_PINS_ENABLE and LCD_PINS_RS.

Two compile-time errors (platformio in VSCode), both apparently due to lack of definition for the Melzi-style display pins.

For now, at least, I'm still using the original display for my Ender-5 (original model). I've installed the Creality Silent 32-bit board, V4.2.7, and am attempting to install Marlin.

### Configuration Files

Configurations to enable LCD for Melzi (original Ender-5 display), BLTouch, dimensions appropriate for Ender-5
[Ender-5_Marlin2.0.zip](https://github.com/MarlinFirmware/Marlin/files/5521921/Ender-5_Marlin2.0.zip)

### Steps to Reproduce

If I uncomment the following directive,
`#define LCD_FOR_MELZI`

**Expected behavior:**

Expect to compile with support for display.

**Actual behavior:**

I get the following errors:
```
In file included from Marlin\src\lcd\dogm\marlinui_DOGM.cpp:42:0:
Marlin\src\lcd\dogm\marlinui_DOGM.h:60:36: error: 'LCD_PINS_ENABLE' was not declared in this scope
Marlin\src\lcd\dogm\marlinui_DOGM.h:60:53: error: 'LCD_PINS_RS' was not declared in this scope
```

#### Additional Information
I can see the definitions in the Sanguino (8-bit) pins definition files. The same connector exists on the V4.2.7 for the display, but they are not defined.

LCD_PINS_ENABLE:
https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/pins/sanguino/pins_SANGUINOLOLU_11.h#L164
and/or
https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h#L67

LCD_PINS_RS:
https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/pins/sanguino/pins_MELZI_CREALITY.h#L66

ellensp, 2020-11-11:
try adding #define RET6_12864_LCD to your Configuration.h

BigBearBrian, 2020-11-12:
Didn't work.

Searching through the codebase, it looks like that would be used with the CR10_STOCKDISPLAY with the Creality V4 controller. AFAIK, the display included with the Ender-5 is enabled by `#define MELZI_CREALITY`, hence my looking at the `pins_MELZI_CREALITY.h` files in the sanguino (8-bit Creality board) definition.

I was able to get this to compile without the display, I think I'm going to try to flash this and see if I can at least get the printer functioning again. I've got a couple of high-priority projects, if I can get them printed, then I can come back and, if necessary, try to figure out the modifications that probably need to be made to `pins_CREALITY_V427.h` (the v4.2.7 32-bit Creality board).

ellensp, 2020-11-12:
I was being hopeful It would be compatible, but clearly different pin outs...  Sadly there is zero documentation for this display (other than existing code)

ellensp, 2020-11-12:
You say this display is with original Ender-5, but the original ender-5 examples configs has #define CR10_STOCKDISPLAY https://github.com/MarlinFirmware/Configurations/blob/bugfix-2.0.x/config/examples/Creality/Ender-5/CrealityV1/Configuration.h#L2108
But its not unheard of that creality changes stuff mid production.

BigBearBrian, 2020-11-12:
OK, ellensp, I'll try that, but probably not 'til tomorrow.

For right now, disgusted. Managed to compile (without any display), figured out that on the very last line of the Marlin Autobuild config it says that I'll have to copy the file over to an sdcard manually (what happened to the USB serial?! Sigh, different architecture, I'm sure is the reason.) Copied it over, put in the printer, powered on, let it sit for a couple minutes, unpowered and removed.

No display, of course, so tried to connect Octoprint.

Sad face. Timed out. Sigh.

ellensp, 2020-11-12:
Also there is a example ender-5 with CrealityV427 controller.... https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples/Creality/Ender-5/CrealityV427  Just select the version of marlin you have..

BigBearBrian, 2020-11-12:
Decided wth, sleep is for the weak, or something like that. 👍 

Uncommented CR10_STOCKDISPLAY. Promptly got an error stating I needed to define one of two pins (one being the above-mentioned RET6_12864_LCD, so added that in:

#define CR10_STOCKDISPLAY
#if ENABLED(CR10_STOCKDISPLAY)
  #define RET6_12864_LCD
#endif

Compiled. Copied over, nearly #####'ed myself when I saw the following error:
[image]

Reset didn't do anything, so WTH, try Ignore.

et voila', I had a display showing my custom printer name. Wonderful!

Tried connecting Octoprint. No joy. Will probably have to dig to find trouble with the "serial port" (aka, USB).

Tried to home via the display... X/Y homed properly, Z started moving down. So, guess I will need to do more twiddling.

So, thanks for the suggestion, at least it moved me forward some. That's enough for tonight; despite the unnecessarily macho comment at the beginning, sleep is all too important for the decrepit body and brain.

ellensp, I see your comment, I think that's actually where I started on this journey. Thanks, I'll use it to double-check the above little tidbits.

G'night (or g'morning, depending on where on this great spinning globe ye may be....)

ellensp, 2020-11-12:
So original issue is not valid as doesn't have this LCD type. So closing.

closed 2020-11-12 by ellensp