MarlinFirmware/Marlin #20101: [BUG] ColorUI - Commenting #define LCD_TIMEOUT_TO_STATUS has no effect

bkiziuk, 2020-11-11:
### Bug Description

Commenting ```#define LCD_TIMEOUT_TO_STATUS``` directive does not disable the timeout.
The menu screen returns to the status screen after about 15 seconds.

I'm not sure if the root cause is the same, but when using Tramming Wizard, the LCD also returns to status after some inactivity.

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5523825/config.zip)

### Steps to Reproduce

1. Enter menu
2. Wait for 15 seconds

**Expected behavior:**

The menu stays visible indefinitely if the option is commented.

**Actual behavior:**

LCD return to status screen after 15 seconds.

#### Additional Information

Board: MKS Robin Nano 1.2 + TFT35
Firmware: Marlin bugfix-2.0.x 2020-11-11

ellensp, 2020-11-11:
If you don't set LCD_TIMEOUT_TO_STATUS it is set in Conditionals_Post.h 
```
// LCD timeout to status screen default is 15s
#ifndef LCD_TIMEOUT_TO_STATUS
  #define LCD_TIMEOUT_TO_STATUS 15000
#endif
```

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-11 by ellensp

thisiskeithb, 2020-11-11:
Set it to 0 if you want to disable the timeout.

bkiziuk, 2020-11-11:
Many thanks for your answer.
So, it would be nice to just add an appropriate comment to clarify the behavior. I tried to create a commit for it but it went a bit messy.