MarlinFirmware/Marlin #20102: [BUG] ColorUI - Not possible to set PID parameters from LCD screen

bkiziuk, 2020-11-11:
### Bug Description

When trying to manually set PID-E parameters from LCD screen, entered values are not accepted.
Instead, following values are set:
PID-P E1: +001.0
PID-I E1: +000.01
PID-D E1: +001.0

### Configuration Files

[config.zip](https://github.com/MarlinFirmware/Marlin/files/5523890/config.zip)

### Steps to Reproduce

1. Set some value to PID-P E1, PID-I E1 or PID-D E1
2. Press Ok to accept value
3. Check the screen for actual value

**Expected behavior:**

The actual value should be equal to the accepted value.

**Actual behavior:**

Following values are used:
PID-P E1: +001.0
PID-I E1: +000.01
PID-D E1: +001.0

#### Additional Information

Board: MKS Robin Nano 1.2
Firmware: Marlin bugfix-2.0.x 2020-11-11

rhapsodyv, 2020-11-20:
We had a bug with `SINGLE_TOUCH_NAVIGATION` that might be causing your issue. Can you test if this fixes your issue too? #20213

bkiziuk, 2020-11-20:
Tested with bugfix-2.0.x 2020-11-20.
Issue is solved.

boelle, 2020-11-20:
@bkiziuk if solved then click green close button below

bkiziuk, 2020-11-20:
@boelle Thought that it needs to merged first ;-) But no problem here :-)

closed 2020-11-20 by bkiziuk

sjasonsmith, 2020-11-20:
@bkiziuk merged into the bugfix branch yes. We don’t wait for them to be tagged in a versioned release to close them.