MarlinFirmware/Marlin #20107: [BUG] not stopping when z endstop is trigerret during M48 test

sogrom, 2020-11-11:
1. Tested with the `bugfix-2.0.x`.

### Bug Description

During M48 probe test, Z axis is not stopping when Z end-stop switch is triggered. Z axis stops only when ABL probe is triggered. 

This may cause damage of head/gantry when there is something wrong with ABL probe.

I'm using BLTouch for ABL, Z endstop for homing...and presumably for damage prevention.

### Configuration Files

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5526205/Marlin.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. start M48 probe test
2. during down movement of Z axis, trigger Z end-stop before probe triggers
 or
1. disconnect ABL probe
2. start M48 

**Expected behavior:**
Z endstop should always stop Z axis and prevent printer damage.

**Actual behavior:**

Gantry crashes in to the bed.

sogrom, 2020-11-11:
enabling #define ENDSTOPS_ALWAYS_ON_DEFAULT - does not fix the issue
sending M120(Enable Endstops) before M48 - does not fix the issue

shitcreek, 2020-11-14:
Looking at the M48 codes, I see it calls the probing codes: probe_at_point --> run_z_probe --> probe_down_to_z  --> probe_down_to_z

 In `probe.cpp` we have this bit of code from line 459 for the `probe_down_to_z` function that checks the trigger state:
```
// Check to see if the probe was triggered
  const bool probe_triggered =
    #if BOTH(DELTA, SENSORLESS_PROBING)
      endstops.trigger_state() & (_BV(X_MIN) | _BV(Y_MIN) | _BV(Z_MIN))
    #else
      TEST(endstops.trigger_state(), TERN(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, Z_MIN, Z_MIN_PROBE))
    #endif
  ;
```
Here we see that it checks only either the `Z_MIN` or the `Z_MIN_PROBE` if you are using `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN`. With the following modification, `ENDSTOPS_ALWAYS_ON_DEFAULT` should now create the special case for either `Z_MIN` or the `Z_MIN_PROBE`.
```
// Check to see if the probe was triggered
  const bool probe_triggered =
    #if BOTH(DELTA, SENSORLESS_PROBING)
      endstops.trigger_state() & (_BV(X_MIN) | _BV(Y_MIN) | _BV(Z_MIN))
    #elif BOTH(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, ENDSTOPS_ALWAYS_ON_DEFAULT)
      EITHER(TEST(endstops.trigger_state(), Z_MIN),  TEST(endstops.trigger_state(), Z_MIN_PROBE)))
    #else
      TEST(endstops.trigger_state(), TERN(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN, Z_MIN, Z_MIN_PROBE))
    #endif
  ;
```

I have not test it out, so I'd recommend taking caution if you do try it.

sogrom, 2020-11-15:
I'm not using the Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN ,
 better option for me on skr e3 mini is to use probe pin because this input doesn't have capacitor an it is faster than end stop pin. This is important especially for precise ABL.

I'll try some code modifications in this area.

sogrom, 2020-11-17:
@shitcreek Tried your code with no luck. 

Tried also this:
But z endstop still not working.

````// Check to see if the probe was triggered
  const bool probe_triggered =
    #if BOTH(DELTA, SENSORLESS_PROBING)
      endstops.trigger_state() & (_BV(X_MIN) | _BV(Y_MIN) | _BV(Z_MIN))
    #else
      TEST(endstops.trigger_state(), Z_MIN) | TEST(endstops.trigger_state(), Z_MIN_PROBE)
    #endif
  ;
````
I'm thinking that problem is in the endstop event area.

bosd, 2020-11-21:
I am testing endstops as well.
I've noticed that homing in the +direction is not working. https://github.com/MarlinFirmware/Marlin/issues/20238
The sensorless endstop might be the issue.

While testing my Marlin setups and endstops I've used the same lead screw assembly on multiple drivers.
I've noticed that the (sensorless) endstop was working on the X and Y axis.
However, using the same settings on the Z-axis, does nothing. The Z-axis does not stop on the endstop. (+ or - direction has no influence)
I've not opened a separate issue, as this might be very similar.

boelle, 2020-11-21:
what if you enable sensorless but use normal endstops for testing?

M119 can show status and you can press them with a finger

bosd, 2020-11-22:
I’ve hooked up some endstops on Zmin and Zmax.
It appears that when homing in the - direction, it looks at the Zmax endstop.
On the GTR V1.0 board, the Stallguard is connected to the Zmin input.
Thus it will never trigger.
The inverting is the result of this code in endstops.h

`#define Z_ENDSTOP (Z_HOME_DIR < 0 ? TERN(HOMING_Z_WITH_PROBE, Z_MIN, Z_MIN_PROBE) : Z_MAX)`

As a workaround I could modify this code for my instance, or work with wire jumpers.

Not sure, if this issue is related. I do not want to hijack this thread, so please continue in #20238 .

closed 2020-12-30 by github-actions[bot]