MarlinFirmware/Marlin #20109: [BUG] (LINEAR ADVANCE : set K factor in TFT_COLOR_UI (or other?) - Bad limit to 999)

phcay, 2020-11-11:
# Bug Description

With my screen configured in TFT_COLOR_UI (with bugfix2.0.x [20201101]), the configuration of the K factor of the LINEAR ADVANCE has a limit of 0 to 999. While it should not reach 10. This makes the cursor unusable.
On another printer running 2.0.7.2, with a DOGM / CR10_STOCKDISPLAY screen, you can exceed 10, but DOGM not having a cursor, it is less annoying.

I made modifications in the bugfix2.0.x [20201101] sources which worked for me, but not being familiar with the language, I may have made some mistakes, so I ask you to check the code below before applying it.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

On screen, menu configuration/conf. advanced/filament/advance K

**Expected behavior:**

K limit of 0 to 9.99

**Actual behavior:**

K limit of 0 to 999

#### Additional Information

\Marlin\src\lcd\extui\ui_api.cpp
630-:        planner.extruder_advance_K[extruder - E0] = constrain(value, 0, 999);
630+:        planner.extruder_advance_K[extruder - E0] = constrain(value, 0, 9.99);

\Marlin\src\lcd\menu\menu_advanced.cpp
113-:        EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 999);
113+:        EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 9.99f);
116-:          EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 999);
116+:          EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 9.99f);
590-:      EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 999);
590+:      EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 9.99f);
593-:        EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 999);
593+:        EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 9.99f);

\Marlin\src\lcd\menu\menu_tune.cpp
205-:      EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 999);
205+:      EDIT_ITEM(float42_52, MSG_ADVANCE_K, &planner.extruder_advance_K[0], 0, 9.99f);
208-:        EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 999);
208+:        EDIT_ITEM_N(float42_52, n, MSG_ADVANCE_K_E, &planner.extruder_advance_K[n], 0, 9.99f);

[K_factor_UI_correction.zip](https://github.com/MarlinFirmware/Marlin/files/5526255/K_factor_UI_correction.zip)

phcay, 2020-11-13:
For information, the limit of 0 to 10, is not a limit that I chose. This is a limit present in the rest of the code (setting, eeprom, gcode ...)

sjasonsmith, 2020-11-15:
@phcay I would like to encourage you to post a pull request with your changes. It is easier to get changes in that way, than to convince someone to pull them from an issue report like this.

phcay, 2020-11-16:
Pull Request created : LCD-LinearAdvance-Limit-correction #20155

closed 2020-11-16 by phcay