MarlinFirmware/Marlin #20111: uint32_t Stepper :: advance_isr () - execution of DIR_WAIT_BEFORE () and DIR_WAIT_AFTER () when LA_steps = 0

phcay, 2020-11-11:
It's not a bug, but something in the code that seems strange to me.
In uint32_t Stepper :: advance_isr (), there is the execution of DIR_WAIT_BEFORE () and DIR_WAIT_AFTER () while LA_steps = 0 and therefore no NORM_E_DIR (), nor REV_E_DIR (). I think we can avoid that, except if I missed something in the code.

Here is my proposition :

#if ENABLED(LIN_ADVANCE)

  // Timer interrupt for E. LA_steps is set in the main routine
  uint32_t Stepper::advance_isr() {
    uint32_t interval;

    if (LA_use_advance_lead) {
      if (step_events_completed > decelerate_after && LA_current_adv_steps > LA_final_adv_steps) {
        LA_steps--;
        LA_current_adv_steps--;
        interval = LA_isr_rate;
      }
      else if (step_events_completed < decelerate_after && LA_current_adv_steps < LA_max_adv_steps) {
             //step_events_completed <= (uint32_t)accelerate_until) {
        LA_steps++;
        LA_current_adv_steps++;
        interval = LA_isr_rate;
      }
      else
        interval = LA_isr_rate = LA_ADV_NEVER;
    }
    else
      interval = LA_ADV_NEVER;
    
    if (LA_steps) {//// PhCay proposal

      DIR_WAIT_BEFORE();

      #if ENABLED(MIXING_EXTRUDER)
        // We don't know which steppers will be stepped because LA loop follows,
        // with potentially multiple steps. Set all.
        if (LA_steps > 0)
          MIXER_STEPPER_LOOP(j) NORM_E_DIR(j);
        else if (LA_steps < 0)
          MIXER_STEPPER_LOOP(j) REV_E_DIR(j);
      #else
        if (LA_steps > 0)
          NORM_E_DIR(stepper_extruder);
        else if (LA_steps < 0)
          REV_E_DIR(stepper_extruder);
      #endif

      DIR_WAIT_AFTER();

      //const hal_timer_t added_step_ticks = hal_timer_t(ADDED_STEP_TICKS);

      // Step E stepper if we have steps
      #if ISR_MULTI_STEPS
        bool firstStep = true;
        USING_TIMED_PULSE();
      #endif

      while (LA_steps) {
        #if ISR_MULTI_STEPS
          if (firstStep)
            firstStep = false;
          else
            AWAIT_LOW_PULSE();
        #endif

        // Set the STEP pulse ON
        #if ENABLED(MIXING_EXTRUDER)
          E_STEP_WRITE(mixer.get_next_stepper(), !INVERT_E_STEP_PIN);
        #else
          E_STEP_WRITE(stepper_extruder, !INVERT_E_STEP_PIN);
        #endif

        // Enforce a minimum duration for STEP pulse ON
        #if ISR_PULSE_CONTROL
          START_HIGH_PULSE();
        #endif

        LA_steps < 0 ? ++LA_steps : --LA_steps;

        #if ISR_PULSE_CONTROL
          AWAIT_HIGH_PULSE();
        #endif

        // Set the STEP pulse OFF
        #if ENABLED(MIXING_EXTRUDER)
          E_STEP_WRITE(mixer.get_stepper(), INVERT_E_STEP_PIN);
        #else
          E_STEP_WRITE(stepper_extruder, INVERT_E_STEP_PIN);
        #endif

        // For minimum pulse time wait before looping
        // Just wait for the requested pulse duration
        #if ISR_PULSE_CONTROL
          if (LA_steps) START_LOW_PULSE();
        #endif
      } // LA_steps

    }//// PhC

    return interval;
  }

#endif // LIN_ADVANCE

sjasonsmith, 2020-11-13:
@phcay it's not clear to me what you are changing from looking at the block of code. I suggest you close this issue, and re-open it as a Pull Request, so that your changes can be seen in context. Discussion of whether your changes are correct can continue there.

boelle, 2020-11-13:
@phcay please close this one and open a PR instead

phcay, 2020-11-13:
Yes, I close it.
This will be my first Pull Request ;-) ...

closed 2020-11-13 by phcay