MarlinFirmware/Marlin #20112: [BUG] BTT SKR Mini V1.1 - Error mapping FILAMENT RUNOUT PIN

tiwanacote, 2020-11-11:
### Bug Description, Configuration Files & Steps to Reproduce

I'm using BTT SKR Mini V1.1 motherboard which doesn't has runout filament input conector, so I have re-maped not used X_MAX PIN pins and have defined FIL_RUNOUT_PIN as configuration.h file comment says.

**pins_BTT_SKR_V1_1.h** 
```
#define FIL_RUNOUT_PIN  P1_28
//#define X_MAX_PIN  P1_28      
```
Into **configuration.h** file:

```
#define FILAMENT_RUNOUT_SENSOR

#if ENABLED(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_PIN 
  #define NUM_RUNOUT_SENSORS   1     
  #define FIL_RUNOUT_STATE     LOW   
  #define FIL_RUNOUT_PULLUP          
  //#define FIL_RUNOUT_PULLDOWN     
  #define FILAMENT_RUNOUT_SCRIPT "M600"
#endif
```

This error message is shown while trying to compile
I have tried with others pins and have the same result.
```
Compiling .pio\build\STM32F103RE_btt\src\src\HAL\STM32F1\Servo.cpp.o

In file included from Marlin\src\HAL\STM32F1\../../inc/../core/boards.h:24:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:35,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:

Marlin\src\HAL\STM32F1\../../inc/../core/macros.h:205:60: error: operator '&&' has no right operand
 #define PIN_EXISTS(PN)      (defined(PN##_PIN) && PN##_PIN >= 0)
                                                            ^
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:797:8: note: in expansion of macro 'PIN_EXISTS'
   #if !PIN_EXISTS(FIL_RUNOUT)
        ^~~~~~~~~~

In file included from Marlin\src\HAL\STM32F1\../../inc/../core/boards.h:24:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:35,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\HAL_SPI.cpp:35:

Marlin\src\HAL\STM32F1\../../inc/../core/macros.h:205:60: error: operator '&&' has no right operand
 #define PIN_EXISTS(PN)      (defined(PN##_PIN) && PN##_PIN >= 0)
                                                            ^
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:797:8: note: in expansion of macro 'PIN_EXISTS'
   #if !PIN_EXISTS(FIL_RUNOUT)
        ^~~~~~~~~~

In file included from Marlin\src\HAL\STM32F1\../../inc/../core/boards.h:24:0,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfigPre.h:35,
                 from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:

Marlin\src\HAL\STM32F1\../../inc/../core/macros.h:205:60: error: operator '&&' has no right operand
 #define PIN_EXISTS(PN)      (defined(PN##_PIN) && PN##_PIN >= 0)
                                                            ^
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:797:8: note: in expansion of macro 'PIN_EXISTS'
   #if !PIN_EXISTS(FIL_RUNOUT)
        ^~~~~~~~~~

*** [.pio\build\STM32F103RE_btt\src\src\HAL\STM32F1\HAL.cpp.o] Error 1
*** [.pio\build\STM32F103RE_btt\src\src\HAL\STM32F1\HAL_SPI.cpp.o] Error 1
*** [.pio\build\STM32F103RE_btt\src\src\HAL\STM32F1\Servo.cpp.o] Error 1`
```

Thanks!

ellensp, 2020-11-11:
You defined the pin in pins_BTT_SKR_V1_1.h
```#define FIL_RUNOUT_PIN  P1_28```
but then in Configuration.h you redefine it as nothing
```#define FIL_RUNOUT_PIN```

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-11 by ellensp

tiwanacote, 2020-11-12:
@ellensp, thanks for your response, but that is not the solution.

A) I have alredy tryed defining only **FIL_RUNOUT_PIN P1_28**  into  **pins_BTT_SKR_V1_1.h** but the sanity check requieres **FIL_RUNOUT_PIN** definition into configuration.h as shows _error message "A"._ (Also I have commented sanity check error in order to jump it but it was worst)

B) Also I have tryed define **#define X_MAX_PIN  P1_28**  only in **configuration.h** (Not in pins_BTT_SKR_V1_1.h ) as is suggested in Marlin comment line, but do not work. See _error message "B"_

**Error Message A:**
```
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:41:0,
                 from Marlin\src\HAL\STM32F1\Servo.cpp:25:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:798:6: error: #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
     #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."
```

**Error Message B:**
```
In file included from Marlin\src\inc/../core/boards.h:24:0,
                 from Marlin\src\inc/MarlinConfigPre.h:35,
                 from Marlin\src\inc/MarlinConfig.h:28,
                 from Marlin\src\MarlinCore.h:24,
                 from Marlin\src\MarlinCore.cpp:31:
Marlin\src\feature/runout.h: In static member function 'static void FilamentSensorBase::setup()':
Marlin\src\inc/../../Configuration.h:1351:25: error: 'P1_28' was not declared in this scope
  #define FIL_RUNOUT_PIN P1_28
                         ^
Marlin\src\inc/../core/macros.h:434:24: note: in definition of macro 'EVAL1'
 #define EVAL1(V...)    V
                        ^
Marlin\src\inc/../core/macros.h:433:30: note: in expansion of macro 'EVAL1'
 #define EVAL2(V...)    EVAL1(EVAL1(V))
                              ^~~~~
Marlin\src\inc/../core/macros.h:432:24: note: in expansion of macro 'EVAL2'
 #define EVAL4(V...)    EVAL2(EVAL2(V))

[ . . . ]

Marlin\src\inc/../core/macros.h:423:24: note: in expansion of macro 'EVAL16'
 #define EVAL(V...)     EVAL16(V)
                        ^~~~~~
Marlin\src\inc/../core/macros.h:475:33: note: in expansion of macro 'EVAL'
 #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
                                 ^~~~
Marlin\src\inc/../HAL/./STM32F1/fastio.h:45:33: note: in expansion of macro '_SET_MODE'
 #define SET_INPUT_PULLUP(IO)    _SET_MODE(IO, GPIO_INPUT_PU)
                                 ^~~~~~~~~
Marlin\src\feature/runout.h:153:36: note: in expansion of macro 'SET_INPUT_PULLUP'
         #define INIT_RUNOUT_PIN(P) SET_INPUT_PULLUP(P)
                                    ^~~~~~~~~~~~~~~~
Marlin\src\feature/runout.h:160:31: note: in expansion of macro 'INIT_RUNOUT_PIN'
       #define _INIT_RUNOUT(N) INIT_RUNOUT_PIN(FIL_RUNOUT##N##_PIN);
                               ^~~~~~~~~~~~~~~
Marlin\src\inc/../pins/pins.h:1201:27: note: in expansion of macro 'FIL_RUNOUT_PIN'
   #define FIL_RUNOUT1_PIN FIL_RUNOUT_PIN
                           ^~~~~~~~~~~~~~
Marlin\src\feature/runout.h:160:47: note: in expansion of macro 'FIL_RUNOUT1_PIN'
       #define _INIT_RUNOUT(N) INIT_RUNOUT_PIN(FIL_RUNOUT##N##_PIN);
                                               ^~~~~~~~~~
Marlin\src\inc/../core/macros.h:460:3: note: in expansion of macro '_INIT_RUNOUT'
   _RPT_OP(_RPT_I)                                             \
   ^~~~~~~
Marlin\src\inc/../core/macros.h:475:38: note: in expansion of macro '_REPEAT'
 #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
                                      ^~~~~~~
Marlin\src\feature/runout.h:161:7: note: in expansion of macro 'REPEAT_S'
       REPEAT_S(1, INCREMENT(NUM_RUNOUT_SENSORS), _INIT_RUNOUT)
       ^~~~~~~~
Marlin\src\feature/runout.h: In static member function 'static uint8_t FilamentSensorBase::poll_runout_pins()':
Marlin\src\inc/../../Configuration.h:1351:25: error: 'P1_28' was not declared in this scope
  #define FIL_RUNOUT_PIN P1_28
                         ^
Marlin\src\inc/../core/macros.h:434:24: note: in definition of macro 'EVAL1'
 #define EVAL1(V...)    V
                        ^
Marlin\src\inc/../core/macros.h:433:30: note: in expansion of macro 'EVAL1'
 #define EVAL2(V...)    EVAL1(EVAL1(V))
                              ^~~~~
Marlin\src\inc/../core/macros.h:432:24: note: in expansion of macro 'EVAL2'
 #define EVAL4(V...)    EVAL2(EVAL2(V))

[ . . . ]

                              ^~~~~
Marlin\src\inc/../core/macros.h:423:24: note: in expansion of macro 'EVAL16'
 #define EVAL(V...)     EVAL16(V)
                        ^~~~~~
Marlin\src\inc/../core/macros.h:475:33: note: in expansion of macro 'EVAL'
 #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
                                 ^~~~
Marlin\src\feature/runout.h:168:32: note: in expansion of macro 'READ'
       #define _OR_RUNOUT(N) | (READ(FIL_RUNOUT##N##_PIN) ? _BV((N) - 1) : 0)
                                ^~~~
Marlin\src\inc/../pins/pins.h:1201:27: note: in expansion of macro 'FIL_RUNOUT_PIN'
   #define FIL_RUNOUT1_PIN FIL_RUNOUT_PIN
                           ^~~~~~~~~~~~~~
Marlin\src\feature/runout.h:168:37: note: in expansion of macro 'FIL_RUNOUT1_PIN'
       #define _OR_RUNOUT(N) | (READ(FIL_RUNOUT##N##_PIN) ? _BV((N) - 1) : 0)
                                     ^~~~~~~~~~
Marlin\src\inc/../core/macros.h:460:3: note: in expansion of macro '_OR_RUNOUT'
   _RPT_OP(_RPT_I)                                             \
   ^~~~~~~
Marlin\src\inc/../core/macros.h:475:38: note: in expansion of macro '_REPEAT'
 #define REPEAT_S(S,N,OP)        EVAL(_REPEAT(S,SUB##S(N),OP))
                                      ^~~~~~~
Marlin\src\feature/runout.h:169:17: note: in expansion of macro 'REPEAT_S'
       return (0 REPEAT_S(1, INCREMENT(NUM_RUNOUT_SENSORS), _OR_RUNOUT));
                 ^~~~~~~~
*** [.pio\build\STM32F103RE_btt\src\src\MarlinCore.cpp.o] Error 1
```

ellensp, 2020-11-12:
P1_28 is not a valid pin for a BTT_SKR_MINI_V1_1

You show the following code which you copied for FIL_RUNOUT_PIN 
`#define X_MAX_PIN  P1_28`

But this line is only present in the following pin files.
Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h
Marlin/src/pins/lpc1768/pins_MKS_SGEN_L.h
Marlin/src/pins/lpc1769/pins_MKS_SGEN_L_V2.h

In the file Marlin/src/pins/stm32f1/pins_BTT_SKR_MINI_V1_1.h is
`#define X_MAX_PIN PA2`
So you need 
`#define FIL_RUNOUT_PIN PA2`

tiwanacote, 2020-11-12:
Thank you! I got the .bin compiled