MarlinFirmware/Marlin #20114: UTF_FILENAME_SUPPORT causes bootlooping

thisiskeithb, 2020-11-12:
### Bug Description

`UTF_FILENAME_SUPPORT` causes bootlooping.

### Configuration Files

Any default config since `UTF_FILENAME_SUPPORT` is enabled by default, but here's one: https://github.com/thisiskeithb/Marlin/tree/myconfig/Prusa-Mini-SKR-Mini-E3-V2-BMG

### Steps to Reproduce

1. Flash your board with the current `bugfix-2.0.x` which has `UTF_FILENAME_SUPPORT` enabled by default.
2. Watch Marlin bootloop.

**Expected behavior:**

No bootlooping.

**Actual behavior:**

Bootlooping.

LinFor, 2020-11-12:
Hello.
Confirming the bootloop on some configurations.
My Fysetc S6 with mini12864 working fine (https://github.com/LinFor/Marlin/tree/e-bot-s6).
But I try my SKR v1.3 with RRD-FG and it's bootloops (https://github.com/LinFor/Marlin/tree/e-bot-skr).
Seems this is platform- or display-specific. I'll try to investigate this problem now.

sjasonsmith, 2020-11-12:
I'm about to post a PR to fix it.

sjasonsmith, 2020-11-12:
@LinFor I see you were the submitter of the UTF8 change. I didn't actually test the UTF-8 behavior, but my changes are pretty minor and prevent the error.

I'll post the PR and tag you in it. You can then review it and let me know whether you would rather solve it a different way.

closed 2020-11-13 by thisiskeithb