MarlinFirmware/Marlin #20119: Only Servo0 works (MKS Robin Nano V1.2)

macace2000, 2020-11-12:
### Bug Description

Servo0 works correct, Servo1 and higher have no function.
Board: MKS Robin Nano V1.2
Tested Marlin Versions:  The actual Bugfix 2.0x

### Configuration Files

[config only nozzle.zip](https://github.com/MarlinFirmware/Marlin/files/5531270/config.only.nozzle.zip)
[config two servos.zip](https://github.com/MarlinFirmware/Marlin/files/5531271/config.two.servos.zip)

### Steps to Reproduce
1. Step
**Configuration.h:**
switching nozzle feature activate and set to Servo0
disabled BL-Touch.
Numbers of Servos is set to 1
**Pins.h:**
Set the Servo0 to Pin PE5. 
**Result:**
The switching nozzle works correct.

2. Step
**Configuration.h:**
switching Nozzle avtivate and set to Servo1
BL-Touch activate
Numbers of Servos is set to 2
**Pins.h:**
Servo0= PA8 (BL-Touch)
Servo1= PE5 (swiching Nozzle)
**Result:** 
BL-Touch works correct,
The swiching nozzle servo did nothing

sjasonsmith, 2020-11-12:
I assume you are building using the mks_robin_nano35 PlatformIO environment? Could you try with the mks_robin_nano35_stm32 environment? You probably have to be using the bugfix-2.0.x branch for it to be available.

macace2000, 2020-11-14:
Hello, 

I have changed to the mks_robin_nano35_stm32 in PlatformIO
[image]

At the first compile it has stoped with this error
[image]
Then I have changed this in variant.h  (I dont now if this is correct )

#define TIMER_SERVO             TIM2
to
#define TIMER_SERVO             TIM6 
[variant.zip](https://github.com/MarlinFirmware/Marlin/files/5540307/variant.zip)

now the compile has no errors or warning to this point:
[image]

sjasonsmith, 2020-11-15:
@rhapsodyv could you take a look at this? The contents of xpt2046.cpp are missing because `HAS_TFT_XPT2046` is not set.
I am not familiar with these files to know whether his configuration actually needs XPT2046 or not.

sjasonsmith, 2020-11-15:
This is specifically when building using `mks_robin_nano35_stm32`.

macace2000, 2020-11-15:
Hello,

I have disabled for a test  #define TOUCH_SCREEN.
Now the compile don't stop with the mks_robin_nano35_stm32 enviroment.
There are no warnings or errors during the compile, but when i flash this firmware to
the printer it makes the TFT-Update up to 100% and stops.
When I restart the printer in the Display comes only Booting.

When I compile this setup with mks_robin_nano35 enviroment, the printer do the update normal.
When the printer starts there is the normal Screen but without touch funktion.

I think the mks_robin_nano35_stm32 enviroment do not work for the MKS Robin nano V1.2,
but i do some further test with the LCD config.

macace2000, 2020-11-15:
Hello,

I found the solution ! I have used the PIN PE5 for all my tests, this was the Problem.
I have switched to PIN PC7 and now it works.

Compiled with mks_robin_nano35 enviroment.
Thanks for your support !

sjasonsmith, 2020-11-16:
So both servos now work with `mks_robin_nano35`?

macace2000, 2020-11-16:
Yes, both servos works.
There are the three warnings during compile with mks_robin_nano35, but all works.

I have changed variant.h back to the origin values. ( It works also with TIM6)
#define TIMER_SERVO TIM2

The Problem was the used Pin PE5. I don't now why, elecrical it makes no sense for me.
I think there is somewere a Problem with this pin in Marlin.

sjasonsmith, 2020-11-18:
I am assigning this to myself to fix the timer conflict. Finding appropriate defaults is a bit complicated, due to various boards setting different values in their `variant.h` files.

I believe Timer 6/7 are Basic timers, so it is possible they don't work exactly as intended. I _think_ they work fine in this HAL, but I want to verify that before I move anything critical like the STEP or TEMP timer there.

sjasonsmith, 2020-12-09:
#20387 is likely the same conflict

sjasonsmith, 2020-12-22:
This was left open to deal with the timer conflict. I am closing it now, since this is now being addressed in #20387.

closed 2020-12-22 by sjasonsmith