MarlinFirmware/Marlin #20120: [BUG] Stop-Print does not break G4 Pause

FlailAway, 2020-11-12:
Marlin bugfix 2.0.7

### Bug Description
"Stop-Print" does not break G4 pause

### Configuration Files
Cannot see how they matter for this.

### Steps to Reproduce
Include a pause at the end of a print GCode for cool down before hitting a mains "Off" toggle switch lever.

; after the final print-lines
G0 Y+320 X+300 ; move close to the toggle switch
G4 S120 ; pause for 2-minutes for hot-end cool-down fan
G1 Y+340 ; go and poke the switch lever

If I select "Stop Print" while it is within the 120-seconds the "stop" is ignored.

**Expected behavior:**
Break the G4 Pause cycle and let me use it again.

**Actual behavior:**
It makes me wait the 120-seconds for the Pause to complete and then I can use it again.

thisiskeithb, 2020-11-12:
Please follow the Issue Template when filing a bug report. This includes attaching your configs.

FlailAway, 2020-11-12:
Thanks for the comment, but the printers are in a building a quarter mile away and no Internet access. I cannot see how this affects anything. It is a generic issue regarding an oversight for stopping the running GCode. It is not specific to any one machine or setting.

boelle, 2020-11-12:
Configurations, please
Please ZIP up your `Configuration.h` and `Configuration_adv.h` files (as
requested in the Issue template) and drop them into your next reply.
We'll check them over and see if anything is amiss.

> bugfix 2.0.7

there are not such version - only 2.0.7 or bugfix-2.0.x

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

thisiskeithb, 2020-11-12:
> I cannot see how this affects anything. It is a generic issue regarding an oversight for stopping the running GCode. It is not specific to any one machine or setting.

You’d be surprised how often that is not the case.

Closing until configs are attached.

closed 2020-11-12 by thisiskeithb