MarlinFirmware/Marlin #20122: [BUG] Bugfix 2.0 - ABL Subdivision results in part of Mesh Table empty

thierryzoller, 2020-11-12:
Bugfix 2.0 (today)

I enabled subdivision and now end up with a broken mesh table :

`    //
    #define ABL_BILINEAR_SUBDIVISION
    #if ENABLED(ABL_BILINEAR_SUBDIVISION)
      // Number of subdivisions between probe points
      #define BILINEAR_SUBDIVISIONS 3
    #endif
`

```
      0      1      2      3      4
 0 +0.142 +0.052 +0.026 +0.049 +0.071
 1 +0.115 +0.032 -0.005 +0.016 +0.017
 2 +0.156 +0.041 +0.011 +0.019 +0.021
 3 +0.144 +0.039 +0.004 +0.009 +0.029
 4 +0.167 +0.070 +0.034 +0.036 +0.079
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8        9       10       11       12
 0 +0.14247 +0.11009 +0.07771 +0.05242 +0.03713 +0.02893 +0.02615 +0.03004 +0.03935 +0.04866 +0.05617 +0.06367  **=======**
 1 +0.13075 +0.10001 +0.06927 +0.04468 +0.02837 +0.01821 +0.01397 +0.01784 +0.02763 +0.03653 +0.04198 +0.04655  **=======**
 2 +0.11903 +0.08993 +0.06083 +0.03695 +0.01961 +0.00749 +0.00179 +0.00563 +0.01590 +0.02439 +0.02780 +0.02943  **=======**
 3 +0.11495 +0.08577 +0.05659 +0.03241 +0.01439 +0.00137 -0.00512 -0.00164 +0.00836 +0.01615 +0.01805 +0.01772  **=======**
 4 +0.12561 +0.09286 +0.06012 +0.03361 +0.01551 +0.00363 -0.00234 +0.00012 +0.00847 +0.01485 +0.01597 +0.01511  **=======**
 5 +0.14390 +0.10588 +0.06786 +0.03801 +0.02017 +0.01050 +0.00572 +0.00680 +0.01277 +0.01744 +0.01832 +0.01791  **=======**
 6 +0.15622 +0.11472 +0.07322 +0.04116 +0.02347 +0.01522 +0.01114 +0.01105 +0.01513 +0.01865 +0.01985 +0.02050  **=======**
 7 +0.15470 +0.11333 +0.07196 +0.03991 +0.02204 +0.01348 +0.00901 +0.00820 +0.01146 +0.01485 +0.01708 +0.01944  **=======**
 8 +0.14719 +0.10776 +0.06833 +0.03741 +0.01925 +0.00960 +0.00424 +0.00290 +0.00585 +0.00966 +0.01349 +0.01817  **=======**
 9 +0.14372 +0.10610 +0.06849 +0.03866 +0.02032 +0.00975 +0.00364 +0.00179 +0.00438 +0.00864 +0.01420 +0.02143 +0.02865
10 +0.14895 +0.11236 +0.07577 +0.04658 +0.02826 +0.01735 +0.01087 +0.00845 +0.01047 +0.01503 +0.02238 +0.03228 +0.04218
11 +0.15822 +0.12253 +0.08684 +0.05825 +0.04006 +0.02898 +0.02226 +0.01929 +0.02069 +0.02560 +0.03487 +0.04765 +0.06043
12 +0.16748 +0.13269 +0.09791 +0.06993 +0.05186 +0.04060 +0.03366 +0.03014 +0.03092 +0.03616 +0.04737 +0.06302 +0.07868
echo:busy: processing
```

thierryzoller, 2020-11-12:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5532765/Marlin.zip)

thierryzoller, 2020-11-12:
https://github.com/MarlinFirmware/Marlin/issues/15818

thierryzoller, 2020-11-12:
Setting Subdivsion to 2 results in the same issue : 

`Bilinear Leveling Grid:
      0      1      2      3      4
 0 +0.106 +0.047 +0.037 +0.049 +0.077
 1 +0.080 +0.041 -0.007 +0.007 +0.016
 2 +0.101 +0.047 +0.007 +0.012 +0.020
 3 +0.105 +0.044 +0.005 +0.011 +0.030
 4 +0.114 +0.066 +0.041 +0.050 +0.086
Subdivided with CATMULL ROM Leveling Grid:
        0        1        2        3        4        5        6        7        8
 0 +0.10570 +0.07312 +0.04666 +0.03722 +0.03666 +0.04025 +0.04866 +0.06189  =======
 1 +0.09006 +0.06551 +0.04288 +0.02313 +0.01102 +0.01493 +0.02500 +0.03345  =======
 2 +0.08018 +0.06095 +0.04066 +0.01327 -0.00737 -0.00365 +0.00714 +0.01177  =======
 3 +0.08856 +0.06628 +0.04376 +0.01695 -0.00299 -0.00124 +0.00711 +0.01038  =======
 4 +0.10069 +0.07312 +0.04716 +0.02321 +0.00664 +0.00639 +0.01214 +0.01577  =======
 5 +0.10366 +0.07285 +0.04441 +0.02029 +0.00423 +0.00362 +0.00955 +0.01519  =======
 6 +0.10520 +0.07302 +0.04366 +0.01990 +0.00464 +0.00427 +0.01114 +0.01987  =======
 7 +0.10942 +0.07990 +0.05329 +0.03292 +0.02049 +0.02047 +0.02793 +0.04029  =======
 8 +0.11420 +0.08875 +0.06617 +0.05013 +0.04116 +0.04160 +0.04967 +0.06596 +0.08569`

daleckystepan, 2020-11-12:
Related https://github.com/MarlinFirmware/Marlin/issues/20027

ellensp, 2020-11-13:
First thing I notice is the unusual formatting Bilinear Leveling Grid is the original grid data.
Subdivided with CATMULL ROM Leveling Grid is only the data points in-between the original grid data, not the full mesh.

Second thing I notice is  BILINEAR_SUBDIVISIONS... 
So on a 5x5 grid with BILINEAR_SUBDIVISIONS = 3, I believe you should get a 12x12 (ie 0-11) subdivision grid. So the 12 row and column seems to be in error.
Second eg a 5x5 gird with BILINEAR_SUBDIVISIONS = 2, shouldn't you get a 8x8 (ie 0-7) subdivision grid ? once again getting and extra row and column.

Looking deeper...

ellensp, 2020-11-13:
I dont have a machine with a real probe...  so can someone give this simple change a try
in Marlin/src/feature/bedlevel/abl/abl.cpp find 
```
  #define ABL_GRID_POINTS_VIRT_X (GRID_MAX_POINTS_X - 1) * (BILINEAR_SUBDIVISIONS) +1 
  #define ABL_GRID_POINTS_VIRT_Y (GRID_MAX_POINTS_Y - 1) * (BILINEAR_SUBDIVISIONS) +1
```
and change to 
```
  #define ABL_GRID_POINTS_VIRT_X (GRID_MAX_POINTS_X - 1) * (BILINEAR_SUBDIVISIONS)
  #define ABL_GRID_POINTS_VIRT_Y (GRID_MAX_POINTS_Y - 1) * (BILINEAR_SUBDIVISIONS)
```

sjasonsmith, 2020-11-13:
This issue is being closed because it is incomplete. When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

ellensp, 2020-11-13:
I'm look at this.

sjasonsmith, 2020-11-13:
@thierryzoller I wasn't trying to minimize the issue you are reporting. We are trying to increase consistency in bug reporting, and part of that is pushing back when instructions have been ignored, with the hope that the reporter would either update the report and re-file with all the requested information.

The consistency provided by the template is important in a couple ways:
- Contributors can more quickly skim through issues looking for ones where they may be able to help
- When a contributor is interested in helping, it is easier to know how to proceed, with less need for back/forth questions just to get started.
- Helps reduce the abundance of issues that people aren't sure what to do with, hopefully reducing the number of ignored issues.

You already have @ellensp interested here, so you are in good hands.

ellensp, 2020-11-13:
Interestingly when I attempt to recreate this I get 
5:34:53.240 > Bilinear Leveling Grid:
15:34:53.241 >       0      1      2      3      4
15:34:53.242 >  0 +0.106 +0.047 +0.037 +0.049 +0.077
15:34:53.244 >  1 +0.080 +0.041 -0.007 +0.007 +0.016
15:34:53.245 >  2 +0.101 +0.047 +0.007 +0.012 +0.020
15:34:53.247 >  3 +0.105 +0.044 +0.005 +0.011 +0.030
15:34:53.249 >  4 +0.114 +0.066 +0.041 +0.050 +0.086
15:34:53.250 > 
15:34:53.250 > Subdivided with CATMULL ROM Leveling Grid:
15:34:53.252 >         0        1        2        3        4        5        6        7        8        9       10       11       12
15:34:53.257 >  0 +0.10600 +0.08452 +0.06304 +0.04700 +0.03922 +0.03689 +0.03700 +0.03878 +0.04300 +0.04900 +0.05715 +0.06707 +0.07700
15:34:53.263 >  1 +0.09559 +0.07759 +0.05959 +0.04456 +0.03307 +0.02455 +0.02019 +0.02148 +0.02693 +0.03326 +0.03967 +0.04697 +0.05426
15:34:53.269 >  2 +0.08519 +0.07067 +0.05615 +0.04211 +0.02692 +0.01221 +0.00337 +0.00417 +0.01085 +0.01752 +0.02220 +0.02686 +0.03152
15:34:53.274 >  3 +0.08000 +0.06733 +0.05467 +0.04100 +0.02337 +0.00474 -0.00700 -0.00674 +0.00041 +0.00700 +0.01037 +0.01319 +0.01600
15:34:53.280 >  4 +0.08415 +0.07050 +0.05685 +0.04244 +0.02457 +0.00594 -0.00604 -0.00649 -0.00029 +0.00541 +0.00804 +0.01017 +0.01230
15:34:53.286 >  5 +0.09352 +0.07726 +0.06099 +0.04522 +0.02837 +0.01201 +0.00137 +0.00014 +0.00464 +0.00904 +0.01136 +0.01359 +0.01581
15:34:53.292 >  6 +0.10100 +0.08248 +0.06396 +0.04700 +0.03096 +0.01648 +0.00700 +0.00522 +0.00844 +0.01200 +0.01444 +0.01722 +0.02000
15:34:53.297 >  7 +0.10341 +0.08352 +0.06363 +0.04574 +0.02956 +0.01538 +0.00611 +0.00412 +0.00704 +0.01063 +0.01370 +0.01744 +0.02119
15:34:53.303 >  8 +0.10393 +0.08302 +0.06212 +0.04348 +0.02695 +0.01269 +0.00344 +0.00147 +0.00451 +0.00859 +0.01272 +0.01788 +0.02304
15:34:53.309 >  9 +0.10500 +0.08385 +0.06270 +0.04400 +0.02770 +0.01385 +0.00500 +0.00319 +0.00637 +0.01100 +0.01637 +0.02319 +0.03000
15:34:53.315 > 10 +0.10763 +0.08740 +0.06717 +0.04948 +0.03446 +0.02198 +0.01419 +0.01270 +0.01591 +0.02104 +0.02782 +0.03654 +0.04526
15:34:53.321 > 11 +0.11081 +0.09227 +0.07373 +0.05774 +0.04458 +0.03397 +0.02759 +0.02659 +0.02982 +0.03552 +0.04391 +0.05477 +0.06563
15:34:53.327 > 12 +0.11400 +0.09715 +0.08030 +0.06600 +0.05470 +0.04596 +0.04100 +0.04048 +0.04374 +0.05000 +0.06000 +0.07300 +0.08600

Taking a closer look at your config, seeing is something is relevant.

sjasonsmith, 2020-11-13:
I can confirm that I saw the failure on an SKR 1.3. I didn't use thierry's values, just enabled `ABL_BILINEAR_SUBDIVISION` and ran G29. I was configured for the default 3 subdivisions, and exactly the same entries came up as NAN (which I think is what the ====== means, in this context).

sjasonsmith, 2020-11-13:
Unfortunately it seems to be inconsistent. After absolutely seeing the problem on my first G29, it went away after I rebooted and I have not yet seen it again.

ellensp, 2020-11-13:
what about values, is Subdivided Gid 0,0 == Bilinear Grid 0,0?

sjasonsmith, 2020-11-13:
Yes, but they are in the original data as well. They just look different because the prints are rounding to different numbers of decimals. That is going to make any attempt to recreate by restoring mesh values a little imperfect, because the values visible are only approximations of the actual floating point data.

sjasonsmith, 2020-11-13:
I've hit it a few more times now. It seems I can reproduce it if I reset, store, then reload the EEPROM. (M502 M500 M501).
Twice in a row it has been bad on the first G29 after that.

If I then run `M500` followed by `M501`, it will save then reload the mesh from EEPROM, and it now looks fine.

daleckystepan, 2020-11-13:
I can reproduce it on my SKR 1.4 with EEPROM disabled. It seems to be a random. Once it is working correctly, once it is not working (printing ====).

thierryzoller, 2020-11-13:
> This issue is being closed because it is incomplete. When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

I believe the question should be whether enough information is provided to replicate the bug. Regardless of form (unless you do automatic parsing). In this case, the only thing missing are the instructions, I thought you'd figure G28, G29 by yourself.

ellensp, 2020-11-13:
Using with bugfix and enabling only the following, build and upload firmware

#define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO
#define SERIAL_PORT -1
#define PROBE_MANUALLY
#define AUTO_BED_LEVELING_BILINEAR
#define GRID_MAX_POINTS_X 2
#define ABL_BILINEAR_SUBDIVISION
#define LCD_BED_LEVELING
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

testing  

Power on controller
connect up usb and a serial monitor

Using LCD
Motion|Bed Leveling|level bed and follow the menu and enter 4 random points of data

You get the error every time (without having to re-probe for real)

eg 
```
 Grid probing done.
 Bilinear Leveling Grid:
       0      1
  0 +0.000 +0.025
  1 +0.000 -0.025
 
 Subdivided with CATMULL ROM Leveling Grid:
         0        1        2        3
  0 +0.00000 +0.00833 +0.01667  =======
  1 +0.00000 +0.00278 +0.00556  =======
  2 +0.00000 -0.00278 -0.00556  =======
  3 +0.00000 -0.00833 -0.01667 -0.02500
```

sjasonsmith, 2020-11-13:
As far as I can tell right now, the interpolation mechanism has to extrapolate past the edge of the mesh. In this process, it is reading past array boundaries in the `bed_level_virt_coord` function in `abl.cpp`.

As an example, I added a print to verify the function  is returning `nan` from `z_values[6][4]` even though the max valid index for that array is `z_values[4][4]`.

I don't have time to pursue it any further right now, but maybe that will point somebody toward the root cause. There are a bunch of possible exit paths in that function. It extrapolates two points past the mesh. The first seems ok, but I think it is falling through to the wrong path for the second.

ellensp, 2020-11-13:
@thierryzoller you have listed a number of issues. But you never follow the template. It is really not acceptable behavior for someone who knows better.  
Having to dig out the information from trial and error using the few bread crumbs you provide in a non standard way is not the way to entice someone to give up their time to help you. 
Remember we are volunteers trying to help you with your issue. So perhaps you could attempt to help us little by filling out the template as requested in the future.

thierryzoller, 2020-11-13:
@ellensp: Apart from the seemingly personal vendetta, which I am going to ignore. Can you substantiate your claims ? Can you explain, what information you needed to "dig out" by trial and error that is not included already in my issue?.  Debugging/Verification requires trial and error and that's hardly because of me.

thierryzoller, 2020-11-13:
https://github.com/MarlinFirmware/Marlin/blob/ba2939b1dc55c04f879ce96e9af990287b4468d2/Marlin/src/feature/bedlevel/abl/abl.cpp#L162

sjasonsmith, 2020-11-13:
@thierryzoller I suggest if you want to discuss head over to Discord. There’s a real bug here, and I’d hate for it to turn into another one of those bugs with 200 off topic replies that never gets fixed.

sjasonsmith, 2020-11-14:
At this point I believe that I have the issue pinpointed. I will hopefully have a fix ready tomorrow.

sjasonsmith, 2020-11-15:
@thierryzoller could you please test the change in the PR I just posted, and reply to the PR with your result?
https://github.com/MarlinFirmware/Marlin/pull/20143

sjasonsmith, 2020-11-18:
This change has been merged. I'll go ahead and close this since you tested it in the PR.

closed 2020-11-18 by sjasonsmith