MarlinFirmware/Marlin #20125: HT-ntc Type Hot End Thermister / Model: HT-NTC100

scatterbomb, 2020-11-13:
### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

scatterbomb, 2020-11-13:
[https://www.aliexpress.com/snapshot/0.html?spm=a2g0s.9042647.6.2.59a44c4d7Bi0g5&orderId=3007198785072639&productId=32890058144](url)
Please add to Temperature Settings/Temperature Sensors available
HT-ntc Type Hot End Thermister / Model: HT-NTC100  Rated up to 320 degrees Thermistor Model: B3950
 For  PT100 E3D V6 Heat Block Cyclops+ and Chimera+ heater hot end.

sjasonsmith, 2020-11-13:
Closing, as the template was completely ignored and even the provided link is broken.

closed 2020-11-13 by sjasonsmith