MarlinFirmware/Marlin #20126: [BUG & Fix] EEPROM size error & I2C port setting

jaeryang-sindoh, 2020-11-13:
Tested
### Bug Description

EEPROM size error

If I use external eeprom(not flash eeprom emulation), eeprom size error is occurred.
Because of size error, default values are loaded when rebooting.
i.e settings.reset() is called when reboot.

There is no I2C port setting for external eeprom

### Steps to Reproduce

1. enable I2C_EEPROM
2. compile and download
3. attach external eeprom

**Expected behavior:**

save values are loaded

**Actual behavior:**

default values are loaded

**Bug fix**

1. add I2C port setting for external eeprom in MarlinCore.cpp

    #if ENABLED(I2C_EEPROM)
    digitalPinHasI2C(PIN_WIRE_SDA);
    digitalPinHasI2C(PIN_WIRE_SCL);
    #endif

2. About EEPROM data size error, size of data structure and one of real data of eeprom is different.

   So two for loop have to be changed.

   1) for loop in settings.cpp line 1570
    // MBL is disabled - skip the stored data
    for (uint16_t q = mesh_num_x * mesh_num_y; q--;) EEPROM_READ(dummyf);

    this for loop is changed to
    for (uint16_t q = GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y; q--;) EEPROM_READ(dummyf);

    2) for loop in settings.cpp line 1619
    // Skip past disabled (or stale) Bilinear Grid data
    xy_pos_t bgs, bs;
    EEPROM_READ(bgs);
    EEPROM_READ(bs);
    for (uint16_t q = grid_max_x * grid_max_y; q--;) EEPROM_READ(dummyf);

    this for loop is changed to
    for (uint16_t q = GRID_MAX_POINTS_X * GRID_MAX_POINTS_Y; q--;) EEPROM_READ(dummyf);

#### Additional Information

Some instruction is not enough.
(ex.https://www.instructables.com/BigTreeTech-SKR-Pro-V11-Adding-a-EEPROM/
https://www.makenprint.uk/3d-printing/3d-printing-guides/skr-v1-4-i2c-eeprom/ )

sjasonsmith, 2020-11-13:
@jaeryang-sindoh can you attach the zip file with your configurations for bugfix-2.0.x? Be sure to include the pins file if you've modified it.

> external eeprom(not flash eeprom emulation), eeprom size error is occurred.

So it works fine with flash emulation? It might help to provide both working configs (with flash EEPROM emulation) and a failing version with the external I2C EEPROM.

zarpyj, 2020-11-16:
Would be nice from you Jaeryang if you add a copy of your file MarlinCore.cpp, honestly, I dont no where to ass this lines
Thank's a lot
JJ

sjasonsmith, 2020-11-18:
Closing, since I don't think we can approach this issue without configuration and pins files.

closed 2020-11-18 by sjasonsmith