MarlinFirmware/Marlin #20128: [BUG] Z Homing with BLTouch bug when size of the bed is too big

matpop38, 2020-11-13:
**Description**

BLtouch 3.0 units not working with marlin on homing. Probe responds to the commands of TFT35 touch screen for deployment, stowing, and shows no errors, but when it comes to homing the Z axis (after X and Y homing) the probe is placed in the middle of the bed, Z axis doesn’t move and the probe doesn't deploy (the printer is in safe mode, no error message I have to reset it to do anything).
Tried on two different 3.0 probes, on two different BTT SKR 1.4 turbo. Tried on an old and the latest one marlin builds (2.0.5.3 and 2.0.7.2) and I obtained the same issue.
The printer is a DIY type printer (looks like a CNC), the dimensions of the bed are 1mx1m and the Z-axis has a stroke of about 40mm.
I noticed that if the size of the bed in configuration.h was less than 550x550cm the homing with the bltouch worked correctly, but above 700x700cm the problem on the Z-homing appears.

**Steps to Reproduce**

1.	Install BL Touch 3.0 on SKR 1.4 turbo board with TFT35 screen
2.	Self test is OK and no flashing lights or errors.
3.	The commands to control the bltouch from the TFT35 screen work
4.	Start homing from the TFT35 screen.
5.	If bed X_BED_SIZE or Y_BED_SIZE <550 it is OK
6.	If bed X_BED_SIZE or Y_BED_SIZE >700 -> Z homing BUG

**Expected behavior:**
Probe should hit bed and be sucked up, triggering the Z endstop.

**Actual behavior: [What actually happens]**
If bed X_BED_SIZE or Y_BED_SIZE >700, Z doesn’t move and probe doesn’t deploy. The probe remains glowing red at the center of the bed, and the printer goes into security (I can’t do anything before reset).

**Additional Information**
This is on the latest build of marlin.
Tested with two new different 3.0 probes, and on two new different SKR 1.4 turbo boards.
No wiring connection issues.
If I lower DEFAULT_AXIS_STEPS_PER_UNIT I can increase a little more bed size before the issue appends.
_Another issue may be related to :_ When the printer works correctly (bed size < 550x550) if I try to print pieces larger than 10 cm with a gcode from cura the printing stops after a few minutes. (With gcode from simplify3D it doesn't seem to be the case).

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/5536458/Configuration.files.zip)

ellensp, 2020-11-13:
owch... `#define DEFAULT_AXIS_STEPS_PER_UNIT   { 6400, 6400, 6400, 6111 }`
so thats 700*6400 steps 4,480,000 steps to get to 700mm.

Can you try (just as a test) temporarily reducing the steps/mm to more traditional numbers link 80,80,6400,6111
See if it correctly deploys the probe (though obviously in the wrong physical place)

boelle, 2020-11-13:
> This is on the latest build of marlin.

And what is that?

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

sjasonsmith, 2020-11-13:
I suspect some math involved in probing is probably overflowing an integer with the larger bed.

matpop38, 2020-11-13:
@boelle? sorry this is an error when I wrote my description. I fixed this.
Thank you, I will test this version in a couple of minutes and tell you if it works.

matpop38, 2020-11-13:
@sjasonsmith, It's possible 🤔, do you think it can be corrected if boelle's version (bugfix-2.0.x branch) doesn't work?

sjasonsmith, 2020-11-13:
I will probably see if I can reproduce the behavior in a simulated environment later today. As long as it can be reproduced I'm sure it could be fixed.

matpop38, 2020-11-14:
Thank you, I hope you will be able to reproduce it because the corrected version of @boelle didn't solve the problem, so I'm still looking for a solution. I can add that I find, by dichotomy, the size limit for which the problem appears (for my configuration of DEFAULT_AXIS_STEPS_PER_UNIT), it's 588mm.

matpop38, 2020-11-14:
I succeeded to increase this limit to 624mm by changing Z_PROBE_LOW_POINT from -1 to -2. However, after this limit, this variable has no more influence.

sjasonsmith, 2020-11-15:
`Z_PROBE_LOW_POINT` should only matter after the probe is already moving. I thought the problem was that the printer never even attempted to probe?

sjasonsmith, 2020-11-15:
This worked fine in the simulated environment, although that clearly doesn't perfectly mimic real hardware.

Here are some instructions to help gather additional information, which might help understand your issue better. I have crossed out a few lines which are probably not important, since this happens during the G28 command.

- Download Marlin bugfix-2.0.x to test with the latest code.
- Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send M502 and M500 to ensure your Configurations are applied.
- Issue the command M111 S247 to enable maximum logging.
- Perform a G28 to do your standard homing procedure.
- ~~Do a G29 to probe the bed. This will also enable bed leveling.~~
- ~~Do some of the moves that revealed problems before. Take notes.~~
- Copy the log output into a .TXT file and attach it to your next reply.
- ~~Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.~~

matpop38, 2020-11-17:
The printer doesn't try to probe if the size is larger than 588mmx588mm, but I don't know why, Z_PROBE_LOW_POINT has an influence on this and changes the limit from 588mm to 624mm.

Ok too bad the simulation doesn't work :( thank you anyway for trying.
Thank you for this detailed tutorial however I won't be able to access the printer before this weekend. I'll keep you informed on Saturday.
In the meantime I found a temporary solution, I divided the number MICROSTEPS by 2 in Configuration_adv.h which allowed me to divide by 2 DEFAULT_AXIS_STEPS_PER_UNIT for X and Y. The printer then probes normally whatever the size of the bed.

matpop38, 2020-11-23:
I come back to you, I made the manipulations that @sjasonsmith advised me, here is the text file.
[BUG printer.txt](https://github.com/MarlinFirmware/Marlin/files/5583486/BUG.printer.txt)

closed 2021-01-27 by github-actions[bot]