MarlinFirmware/Marlin #20129: [BUG] Possible overflow in planner.cpp when using BLOCK_BUFFER_SIZE higher then 64

FanDjango, 2020-11-13:
When defining a `BLOCK_BUFFER_SIZE` higher than `64` in `configuration_adv.h` **_in combination with_** using `#define DISABLE_INACTIVE_EXTRUDER` in `configuration.h`, there will be a compiler warning (with possibly unwanted consequeces) as follows:

```
  438 |       static uint8_t g_uc_extruder_last_move[EXTRUDERS];
      |                      ^~~~~~~~~~~~~~~~~~~~~~~
In file included from Marlin\src\module\../inc/../core/boards.h:24,
                 from Marlin\src\module\../inc/MarlinConfigPre.h:37,
                 from Marlin\src\module\../inc/MarlinConfig.h:28,
                 from Marlin\src\module\../MarlinCore.h:24,
                 from Marlin\src\module\planner.h:33,
                 from Marlin\src\module\planner.cpp:65:
Marlin\src\module\planner.cpp: In static member function 'static bool Planner::_populate_block(block_t*, bool, const abce_long_t&, const xyze_pos_t&, feedRate_t, uint8_t, const float&)':
Marlin\src\module\planner.cpp:2052:62: warning: unsigned conversion from 'int' to 'uint8_t' {aka 'unsigned char'} 
changes value from '256' to '0' [-Woverflow]
 2052 |             g_uc_extruder_last_move[N] = (BLOCK_BUFFER_SIZE) * 2; \
      |                                          ~~~~~~~~~~~~~~~~~~~~^~~
```

Changing the definition of `g_uc_extruder_last_move` from `uint8_t` to `uint16_t` in `planner.cpp/planner.h` would solve this and this won't cost very much memory (plus one byte per extruder).

boelle, 2020-11-13:
please submit the fix as a PR, then it will be included and this is fixed faster

FanDjango, 2020-11-13:
PR #20130

closed 2020-11-13 by FanDjango