MarlinFirmware/Marlin #20132: LPC1768 and LPC1769 will not compile when trying to use Adafruit MAX31865 board & PT100 sensor

GadgetAngel, 2020-11-13:
### Bug Description

When I compile Marlin for SKR PRO V1.x board and use TEMP_SENSOR_0 set to -5, I can compile without a problem because in .vscode\c_cpp_properties.json file for SKR PRO V1.x boards it includes an ARDUINO constant set for 10808. This ARDUINO constant is used in Adafruit MAX31865::Adafruit_MAX31865.h file which in turn is used by Marlin's temperature.cpp file. 

The PROBLEM: If **ARDUINO constant is not set** then Adafruit_MAX31865.h will include WProgram.h which **is not** what we want for the **LPC1768 and LPC769 environments**.  The Adafruit_MAX31865.h file has the following code:
```
#if (ARDUINO >= 100)
#include "Arduino.h"
#else
#include "WProgram.h"
#endif 
```
I am sure the above code was included to allow 8-bit boards to access the Adafruit MAX31865 library.  **But**, it look like a problem arose and the .vscode\c_cpp_properties.json file for (SKR V1.3, SKR V1.4, SKR V1.4 TURBO,etc) LPC1768 and LPC1769 environments only have ARDUINOLPC constant set to 10805.  The ARDUINO constant is NOT set for LPC1768 and LPC1769 envirnoments. In fact, the ARDUINO constant is NULL when the code hits the Adafruit_MAX31865.h file. 

So when Marlin compiles and the compiler hits the include for the Adafruit MAX31865 library in temperature.cpp the Marlin code will load WProgram.h file instead of loading the correct Arduino.h file.

This means non of the LPC1768 or LPC1769 can use the PT100 with Adafruit MAX31865 boards on either TEMP_SENSOR_0 or TEMP_SENSOR_1.  In other words, LPC1768 and LPC1769 can not use TEMP_SENSOR_0 set to -5 or TEMP_SENSOR_1 set to -5.

This bug report applies to both the release branch of Marlin and the bugfix-2.0.x branch of Marlin!

### Configuration Files

Configuration files for LPC1768 env:
[Configuration_files_LPC1768.zip](https://github.com/MarlinFirmware/Marlin/files/5538516/Configuration_files_LPC1768.zip)

Configuration files for LPC1769 env:
[Configuration_files_LPC1769.zip](https://github.com/MarlinFirmware/Marlin/files/5538518/Configuration_files_LPC1769.zip)

In the above two zip files I also included the Adafruit_MAX31865.h file I changed to temporarily get around the problem:
Since Marlin defines the ARDUINOLPC constant instead of ARDUINO constant that the Adafruit Library wants I added code to the Adafruit_MAX31865.h header file :
```
#if (ARDUINO >= 100 || ARDUINOLPC >= 100)
#include "Arduino.h"
#else
#include "WProgram.h"
#endif 
```
I know this is a third party library.  But the above fix allows the LPC1768 and LPC1769 environments to now compile without issue when they have TEMP_SENSOR_0 set to -5 in configuration.h file.

Here are the .vscode\c_cpp_properties.json files from LPC1768, LPC1769 and BIGTREE_SKR_PRO environments for your consideration:
[c_cpp_properties.json files.zip](https://github.com/MarlinFirmware/Marlin/files/5538672/c_cpp_properties.json.files.zip)

### Steps to Reproduce

1. Download either Marlin 2.0.7.2 release branch or the current Marlin bugfix-2.0.x branch
2. Setup platformio.ini to use either LPC1768 or LPC1769, in configuration.h set the following:
```
#define TEMP_SENSOR_0 -5
#define MOTHERBOARD BOARD_BTT_SKR_V1_4_TURBO for LPC1769
or
#define MOTHERBOARD BOARD_BTT_SKR_V1_4 for LPC1768
```
3. in the pins_BTT_SKR_common.h:
```
#define TEMP_0_PIN                    P0_23_A0
#ifndef MAX6675_SS_PIN
  #define MAX6675_SS_PIN              TEMP_0_PIN
  #define MAX31865_CS_PIN             MAX6675_SS_PIN
#endif 

Try to compile the code and you will get the following errors (below shows the LPC1769 environment compiler output):
Executing task: C:\Users\joann\.platformio\penv\Scripts\pio.exe run <

Processing LPC1769 (platform: https://github.com/p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.3.zip; board: nxp_lpc1769; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/nxplpc-arduino-lpc176x/nxp_lpc1769.html
PLATFORM: NXP Arduino LPC176x (0.1.3) > NXP LPC1769
HARDWARE: LPC1769 120MHz, 31.97KB RAM, 464KB Flash
DEBUG: Current (cmsis-dap) On-board (cmsis-dap) External (blackmagic, jlink)
PACKAGES:
[... 901 lines cut ...]
Compiling .pio\build\LPC1769\FrameworkArduino\HardwareSerial3.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\Print.cpp.o
Archiving .pio\build\LPC1769\libCMSIS.a
Compiling .pio\build\LPC1769\FrameworkArduino\SoftwareSPI.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\SoftwareSerial.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\Stream.cpp.o
Archiving .pio\build\LPC1769\liblpc176x.a
Compiling .pio\build\LPC1769\FrameworkArduino\Tone.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\WInterrupts.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\Wire.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\arduino.cpp.o
Compiling .pio\build\LPC1769\FrameworkArduino\main.cpp.o
Archiving .pio\build\LPC1769\libFrameworkArduino.a
Linking .pio\build\LPC1769\firmware.elf
Checking size .pio\build\LPC1769\firmware.elf
Building .pio\build\LPC1769\firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  40.9% (used 13376 bytes from 32736 bytes)
Flash: [===       ]  33.5% (used 159176 bytes from 475136 bytes)
================================================================================ [SUCCESS] Took 62.15 seconds ================================================================================
Environment    Status    Duration
-------------  --------  ------------
LPC1769        SUCCESS   00:01:02.146
================================================================================ 1 succeeded in 00:01:02.146 ================================================================================ 
```

p3p, 2020-11-13:
This is actually expected behaviour, at least from my lpc176x frameworks point of view, it is not a standard compliant Arduino framework implementation so I don't advertise it as such by setting the `ARDUINO` definition. 

Some general purpose Arduino libraries have been tested and/or modified to work with my framework but in general they don't tend to be supported.

Not sure what the correct solution for Marlin would be in this case, other than making the framework actually Arduino compliant.

GadgetAngel, 2020-11-13:
@p3p I was not aware that LPC1768 and LPC1769 were not meant to run Adafruit MAX31865 boards. 

Tomorrow, I will actual bench test my above suggestion just to see if a  SKR V1.3 or SKR V1.4 or SKR 1.4 Turbo will work with the Adafruit MAX31865 library.  

I will hook up an Adafruit MAX31865 with PT100 sensor via hardware spi to these boards one at a time. After I run the test, I will report my results back here.

Let me run the test to see if this hack will allow SKR V1.3 or SKR V14 or SKR V1.4 Turbo boards to work with the Adafruit MAX31865 library.   So please leave this ticket open until I can document whether or not these boards from Bigtreetech can or can not work with Adafruit MAX31865. 

Thank you for responding so quickly.

GadgetAngel, 2020-11-14:
@p3p I am still testing, please leave this open until I have told you I have given up trying. Thanks for your patience

GadgetAngel, 2020-11-16:
@p3p I have a question for you: In Marlin/src/HAL/LPC1768/include/SPI.h file you have the CLASS SPISettings which has a Method called SPISettings.  This is the method I am asking about: 
```
  SPISettings(uint32_t spiRate, int inBitOrder, int inDataMode) {
    init_AlwaysInline(spiRate2Clock(spiRate), inBitOrder, inDataMode, DATA_SIZE_8BIT);
  }
.
.
.
  static inline uint32_t spiRate2Clock(uint32_t spiRate) {
    uint32_t Marlin_speed[7]; // CPSR is always 2
    Marlin_speed[0] = 8333333; //(SCR:  2)  desired: 8,000,000  actual: 8,333,333  +4.2%  SPI_FULL_SPEED
    Marlin_speed[1] = 4166667; //(SCR:  5)  desired: 4,000,000  actual: 4,166,667  +4.2%  SPI_HALF_SPEED
    Marlin_speed[2] = 2083333; //(SCR: 11)  desired: 2,000,000  actual: 2,083,333  +4.2%  SPI_QUARTER_SPEED
    Marlin_speed[3] = 1000000; //(SCR: 24)  desired: 1,000,000  actual: 1,000,000         SPI_EIGHTH_SPEED
    Marlin_speed[4] =  500000; //(SCR: 49)  desired:   500,000  actual:   500,000         SPI_SPEED_5
    Marlin_speed[5] =  250000; //(SCR: 99)  desired:   250,000  actual:   250,000         SPI_SPEED_6
    Marlin_speed[6] =  125000; //(SCR:199)  desired:   125,000  actual:   125,000         Default from HAL.h
    return Marlin_speed[spiRate > 6 ? 6 : spiRate];
  } 
```
This method or routine is called by the Adafruit_MAX31865.cpp code as shown below:
```
static SPISettings max31865_spisettings =
    SPISettings(500000, MSBFIRST, SPI_MODE1);
```
The thing that worries me is that your SPISettings calls spiRate2Clock(spiRate).  The Adafruit_MAX31865.cpp is not aware that Marlin uses index to speed. It just provides the 500,000 as its SPI speed.  So with your code when Adafruit_MAX31865 calls SPISettings the actual speed will be set to 125000 because the spiRate is greater than 6 in the call.  This is 1/4 of the speed that it should be.

I am trying to decide if this is a non issue? I have changed the Adafruit call to SPISettings(4, MSBFIRST, SPI_MODE1); but I have not seen a difference. I am looking for your opinion.  The maximums SPI speed the MAX31865 can do is 5MHz.

I am still testing. I want to see if I can now use TMC2130 in SPI mode and see how that effects my setup.  

If you can get back to me about the above issue I would be grateful.  

Also do you know of a way to get the linker to include the Arduino.h file without having to change the Adafruit_MAX31865.h file? 

I am older than you and I might not be aware of new changes in compiler and linker directives so I am asking. In my time this was not possible.

rhapsodyv, 2020-11-16:
Marlin SPI usages is not very “standard”  right now. We are working on in a better class and interface for that. But it’s a working in progress.
When I did that SPISettings, I really wasn’t aware of MAX31865. I did that class just for compatibility and didn’t pay too much attention for its interface.

I will take a look in your issue.

GadgetAngel, 2020-11-16:
@rhapsodyv and @p3p I can get the SKR V1.4 Turbo to run the Adafruit MAX31865 board in software SPI and Hardware SPI when SDCARD_CONNECTION is set for LCD.  

I have another question: if I change SDCARD_CONNECTION to ONBOARD does the **default hardware SPI bus** change from SPI-0 bus to SPI-1 bus (from "EXP2 connector" to "onboard SD card reader")?

GadgetAngel, 2021-01-12:
> This issue has had no activity in the last 30 days. Please add a reply if you want to keep this issue active, otherwise it will be automatically closed within 7 days.

This issue is still under review because the pull request that fixes this issue is still going through the review process.

GadgetAngel, 2021-02-10:
This issue is resolved because the PR #20447 has now been merged in bugfix branch

closed 2021-02-10 by GadgetAngel