MarlinFirmware/Marlin #20134: #define SDCARD_CONNECTION ONBOARD

voltmeter101, 2020-11-14:
Hello I'm new here, I hope I respect all the rules to describe my problem.

**Description**

when changing the #define SDCARD_CONNECTION ONBOARD when wanting to use the onboard reader instead of the lcd reader i am able to see the files and open the g code from the printers motherboard sd reader but it will not print nothing heats up it just counts but it still sees the lcd sd card reader and will print fine from that one 

Steps to Reproduce

Install a tft35 display on skr 1.4 board
compile marlin to use the onboard reader instead of the lcd
an sd card with actual gcode to be able to print something
and try to actually print from the sd card read instead of looking to see if it reads it 

Expected behavior: I would expect that by changing #define SDCARD_CONNECTION LCD to #define SDCARD_CONNECTION ONBOARD that marlin would be able to see and print from the onboard card reader and not see the lcd sd card reader

Actual behavior: [What actually happens]
upon changing marlin to use the onboard card reader the printer is able to physically see both the lcd and onboard readers and i am able to scroll through the files however when i put a card into the onboard slot and to to print from it the printer doesnt start the g code it just has a timer counting how long its been printing for no heated bed, homing or anything else however when i try to print from the lcd then it prints as normal

Additional Information
Tested with two new different skr 1.4 boards and two different tft35 displays
[confighfiles.zip](https://github.com/MarlinFirmware/Marlin/files/5539867/confighfiles.zip)

shitcreek, 2020-11-14:
Before submitting a bug report, please try the sources listed below to determine if it is indeed a bug. Also you could try searching through the issue queue to see if someone else has brought up a similiar issue and has been resolves.
I've found this one that may pertain to this very issue: https://github.com/MarlinFirmware/Marlin/issues/19243

https://github.com/MarlinFirmware/Marlin/issues/19243#issuecomment-686460684
From: @AnHardt 
> As far as I remember this is not a bug but kind of a feature.
> It's about preventing writes from both sides, not knowing about each other, and potentially corrupting the file system. Since Marlins ancient sd-library can't do file- or directory-locks the used strategy is to make a kind of disk-lock - trying to prevent the different access channels have a chance to access at the same time.
> It's also about to prevent situations like, Marlin is printing a file and the USB-MSD is trying to delete the file/directory.
> 
> Yes - it's crude.

For best results getting help with configuration and troubleshooting, please use the following resources:

MarlinFW.org Marlin Documentation
RepRap.org Marlin Forum
Tom's 3D Forums
Facebook Group "Marlin Firmware"
Facebook Group "Marlin Firmware for 3D Printers"
Marlin Configuration on YouTube
Marlin Discord server. Join link: https://discord.gg/n5NJ59y
After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a bug report.

I will close this issue for now until this is determined to indeed be a bug.

closed 2020-11-14 by shitcreek

voltmeter101, 2020-11-14:
I'm not trying to be ride but i don't know why you guys keep ignoring this issue and closing it out, if it can't be fixed then so be it we are just forced to use the lcd reader,i dont think you understand completely what I'm saying the other person is having a similar case but they are trying to print from a computer normally I would realize that to be the case if my computer were hooked up to the printer. No computer is hooked up to the printer when I'm trying to print. I want to he able to print from both the lcd and motherboard if possible but if only one is possible i would like it to be the motherboard vs the lcd. And the option that tells marlin hey dont use the lcd connection use the onboard connection instead only makes the system able to read from it but it cant print from it. And i get what you are saying about the usb host and if you look at my config file i specifically tell marlin to turn off access for usb host under 

#define NO_SD_HOST_DRIVE which states that it disables the sd access over USB 
So that should completley eliminate the problem where its tring to be a usb host and an sd card thus corrupting the file system. That and the added fact that everyone seems to be having this same problem with the only solution is to use the lcd readers card. 

I also reached out to bigtreetech and they said that it os a bug in marlin and until they can get a fix for it the only work around is to use the lcd

I will go on the site you suggested and try that but i know this has been going on for a while even if you do use the marlin big fix files and it seems to affect all the 1769 and 1768 boards.

ellensp, 2020-11-14:
It not an issue.  

The TFT is s small computer not just a screen + sdcard reader 

  [Controller] <-serial-> [TFT] <-SPI-> [sdcard]

The TFT can always read the sd card on the tft.  You cannot disable it without re writing the TFT firmware supplied from BTT
The TFT when it reads from sd card on TFT it just send the data to Marlin over serial.

re SDCARD_CONNECTION ONBOARD
This tell marlin itself where to look for a directly connected sd card. So it can internally read the file and process the gcode.
It cannot be set to LCD when you use the TFT, as that device only shows its serial port, it does not allow the controller to directly access the sdcard slot. 

NO_SD_HOST_DRIVE is once again in marlin only, it does not extend to the TFT