MarlinFirmware/Marlin #20137: [BUG] Round objects end up in Oval shape after 2nd Layer (SKR Turbo, Y-Axis *2)

thierryzoller, 2020-11-14:
### Bug Description
In a sporadic fashion, the printer doubles the Y movements starting 2nd layer (or sometimes later) resulting in Oval shaped objects. X and E movements seem not affected and by measurement I can determine that it doubles the Y steps/movements.

I understand it will be near impossible to replicate but open this issue for other users that may have the same issue to chime in.

A user on FB said that it is solved by RESETING EEPROM <- 
I can not directly confirm as it happens sporadically.

### Configuration Files
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5540656/Marlin.zip)

- SKR Turbo + TMC2209 in Uart + Sensorless Homing
- Y-ma = 1100
- Sensitivity = 80

### Steps to Reproduce
1. Print file
2. Sometimes it happens sometimes it doesn't.

**Expected behavior:**

Prints a round object

**Actual behavior:**

Prints an Oval objects with Y dimensions doubled.

[image]

Gcode file: [SX1_PLA_MittelRohrHalter 4mm.zip](https://github.com/MarlinFirmware/Marlin/files/5540661/SX1_PLA_MittelRohrHalter.4mm.zip)

thierryzoller, 2020-11-14:
@sjasonsmith @ellensp : Better?

thierryzoller, 2020-11-14:
Addition:
If it happens BEFORE print the BED runs into the ENDstop and it won't stop until you power it off.
There is something wrong with how Y step state is maintained.

thierryzoller, 2020-11-14:
Related? 
https://github.com/MarlinFirmware/Marlin/issues/14361

sjasonsmith, 2020-11-14:
Is all the plastic in the picture from the same print, or is this two prints shown together for comparison?

sjasonsmith, 2020-11-14:
Also, when you say the bed grinds against the endstop, is that while homing or during a different move?

I suspect that the number of micro steps on that driver is not being programmed correctly, or is for some reason lost after initialization. My questions are to help me form a better understanding of the problem to make sure the behavior matches what I would expect if that happened.

thierryzoller, 2020-11-14:
> Is all the plastic in the picture from the same print, or is this two prints shown together for comparison?

Same print, first layer went down perfect, then it suddenly does Y moves doubled the required.

thierryzoller, 2020-11-14:
> Also, when you say the bed grinds against the endstop, is that while homing or during a different move?
Homing (but again, sporadically only)

sjasonsmith, 2020-11-14:
Ok, changing mid-print conflicts with some of my original thoughts.

My current thought is that the Y driver is resetting during the print and goi by back to power-on default settings.

If you have a spare driver, please exchange it and test whether the problem still exists.

If you don’t have a spare try switch the X and Y driver, so see if the problem still remains with Y.

thierryzoller, 2020-11-14:
> If you don’t have a spare try switch the X and Y driver, so see if the problem still remains with Y.
Excellent idea, will do.

thierryzoller, 2020-11-25:
I replaced the stepper and it has since then not reproduced. For the record, out of 12 bigtree tmc2209 - 3 where defective, in some cases like this, difficult to detect.

closed 2020-11-25 by thierryzoller