MarlinFirmware/Marlin #20144: [BUG] bltouch not does not deploy when homing

bryankenote, 2020-11-15:
### Bug Description
Printer: Ender 3
Main Board: BTT SKR Mini E3 v1.2
Firmware: tested on 2.0.5.4, 2.0.7.2, and bugfix-2.0.x

When I first turn on my printer, the probe is deployed and stowed twice with no issues. But my bltouch is not deploying the probe when homing or using the `M401` command.

### Configuration Files
[configs.zip](https://github.com/MarlinFirmware/Marlin/files/5541804/configs.zip)

### Steps to Reproduce

1. Turn on printer
2. Home Z.

**Expected behavior:**
The probe should deploy when homing. 

**Actual behavior:**
The probe does not deploy and the nozzle crashes into the bed.

ellensp, 2020-11-15:
Can you provide pictures on how its wired on board, want to see colors of the wires and positions.

bryankenote, 2020-11-15:
Is this good enough?

[image]

ellensp, 2020-11-15:
Yes that picture is fine.

thisiskeithb, 2020-11-15:
The servo cable (brown/red/yellow) looks to be 180° out:
[image]

ellensp, 2020-11-15:
Agreed, was just typing the same.

bryankenote, 2020-11-15:
Ah, that did it. That should have been the first thing I checked.

Awesome thanks for the help.

closed 2020-11-15 by bryankenote