MarlinFirmware/Marlin #20146: [FR]  Need an ability to load Thermistor calibration tables from SD card

dmderev, 2020-11-15:
### Description

Many Marlin machines have uncalibrated temperature readings. It may be caused by tolerances on the power supply reference voltages or resistors or ADC ranges. The values may drift over time and changed when thermistors are replaced. It does not make sense to edit source code and recompile firmware every time calibration is needed. 
The suggested feature is either to read the calibration table or temperature scaling factor from flash card file and reprogram calibration table in the device. Alternatively, a configuration command may be implemented to replace/patch temperature calibration table from serial interface. Separate calibration options are needed for each temperature sensor (extruders, bed, etc...).

### Feature Workflow

1. [First Action] - User sets the temperature from control panel and independently reads the actual temperature. It may be the endpoint or few calibration points
2. [Second Action] - the table of "displayed" and actual temperatures is created. 
3. The table is copied to the flash file in a human-readable format, eg. CSV.
4. The card is inserted into the machine and read by the bootloader/updater, or alternatively, it is sent to a serial interface,
5. The firmware updates current look up table in flash. If the data points in the table are not matching the knee points of LUT, the software may interpolate and modify the original table accordingly.

#### Additional Information

Alternatively, a command to adjust the scale (or endpoint) of the current calibration curve and save it may be provided. This is easier and may achieve similar results.

thisiskeithb, 2020-11-15:
You can (and should) run a [PID autotune with `M303`](https://marlinfw.org/docs/gcode/M303.html).

ellensp, 2020-11-15:
In my view this is rather against the design philosophy of Marlin. In that hardware changes necessitate re compiling..

The table is a curve. Approximating from a few data points will not be good enough. It is normally generated from a formula.  

"and read by the bootloader/updater"  You would need to code and write a new bootloader. Changing a bootloader normally requires additional hardware and a high level of technical knowledge. This can brick the controller forever, if done wrong..   Not advisable for beginners.

dmderev, 2020-11-15:
IMHO, it is a necessary feature, and from the implementation the configuration look up table may be stored in a dedicated location in the emulated EEPROM in the same way as other configuration data. The alternative method is to include it in the configuration file for which there is already a read/write infrastructure. It is a parameter, not a change in hardware infrastructure which requires recompilation. 
On the other comment: thermistor is the source of data to PID control, its errors will directly effect the actual temperature even with a perfectly tuned PID parameters.
I agree with the point of not rewriting bootloader. So maybe there is another better way. I did not look at how configuration data is stored - maybe it is is a matter of moving the conversion table into configuration_store.cpp as other parameters.

Sebazzz, 2020-11-27:
> In my view this is rather against the design philosophy of Marlin. In that hardware changes necessitate re compiling..

You don't need to. PID values are saved to EEPROM - and *if you want*  you can put them in your Configuration.h.

dmderev, 2020-11-28:
Sebazz, could you elaborate how the calibration table can be placed into the EEPROM without recompilation? And how Configuration.h ends up in the EEPROM - without recompilation? I think that if there is such a method available and since EEPROM apparently can be saved to flash card and restored - it would be very convenient for instance to write a python script to generate binary from the calibration data and put it into EEPROM eventually. The recalibration may be needed when a hotend or thermistor is replaced and might be useful for even verifying the off-the-shelf settings. Since the thermistors are very non-linear (especially in the simple connection as in many boards), the newly purchased machines are usually 15-20C off...

Sebazzz, 2020-11-28:
> Sebazz, could you elaborate how the calibration table can be placed into the EEPROM without recompilation? And how Configuration.h ends up in the EEPROM - without recompilation?

In MarlinSettings::load:

[image]