MarlinFirmware/Marlin #20149: FILAMENT_RUNOUT_SCRIPT executed only one time

easycorexy, 2020-11-15:
Hi,
I have the following issue:

### Bug Description

The FILAMENT_RUNOUT_SCRIPT is executed only one time, if the filament runs out again the script won't be executed. It won't be executed also in the next print jobs. I order to work again, the printer has to shutted down and than it works again, but only one time. 
it doesn't matter if i print from SD or from Host

### Configuration Files

[Marlin_issue_FILAMENT_RUNOUT_SCRIPT.zip](https://github.com/MarlinFirmware/Marlin/files/5541964/Marlin_issue_FILAMENT_RUNOUT_SCRIPT.zip)

### Steps to Reproduce

1. put something o execute in #define FILAMENT_RUNOUT_SCRIPT
2. remove filament (firt time it works just fine)
3. remove the filament again after some time (now it doesn't work anymore)

**Expected behavior:**

I expect that FILAMENT_RUNOUT_SCRIPT to be executed every time when the filament runs out

**Actual behavior:**

the FILAMENT_RUNOUT_SCRIPT is executed only for the first filament runout the for the next filamnt runout events is not executet. In order to be executed again, the printer has to be shutted off an than on.

sjasonsmith, 2020-11-15:
Please test the [bugfix-2.0.x branch](https://github.com/MarlinFirmware/Marlin/archive/bugfix-2.0.x.zip) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

If it still fails with the bugfix branch, please post new configuration files that work with the `bugfix-2.0.x` branch. Be sure to include anything else you may have modified, such as your pins file.

rhapsodyv, 2020-11-21:
I tested it in the simulator, and I could change the filament two times with success.

rhapsodyv, 2020-11-21:
Now I found the issue. The user are using a totally custom script. The problem is: only pause gcode resets the state of filament runout. And we can't reset the state of runout in every place. 

To use a custom script, the user should send `M412 R` in the end of his custom gcode, to reset the state of runout code. This will solve your problem.

https://marlinfw.org/docs/gcode/M412.html

closed 2020-11-21 by rhapsodyv