MarlinFirmware/Marlin #20152: [BUG] SKR 1.4 Turbo E0 INVERT_E0_DIR does not affect anything

dkowis, 2020-11-15:
### Bug Description

Switching INVERT_E0_DIR does not change the direction of the Extruder. It will retract and extrude just fine (and seemingly the right amount), but it's doing it in the opposite direction, because I've switched to a Geared Extruder.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5542848/Configuration.zip)
Here's the zip file, but it is available on my github as well: https://github.com/dkowis/Marlin

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Have geared extruder
2. `G1 E10`
3. Watch extruder retract filament instead of extrude...
4. Flip `INVERT_E0_DIR` from true to false
5. Recompile, reflash firmware
6. `G1 E10`
7. Watch extruder retract 10mm still :(

**Expected behavior:**

I'd expect toggling `INVERT_E0_DIR` to make my extruder change directions

**Actual behavior:**

Toggling `INVERT_E0_DIR` doesn't affect my extruder direction at all.

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

thisiskeithb, 2020-11-15:
After flashing, be sure to run `M502` followed by `M500`.

Does filament move the correct direction when adjusting Motion -> Move Axis -> Extruder?

dkowis, 2020-11-15:
Yes, I have done that. 

When I move the filament via the LCD (Motion -> Move Axis -> Extruder) I have to move in the negative direction to get filament to extrude. Positive direction moves retract filament. Same as the `G1 E*whatev*` command.

dkowis, 2020-11-15:
Okay, after suggestions on Discord, I have tried again, and this time it does in fact do the right things. I must've missed a step when doing it myself lots and lots of times :(

This doesn't seem to be a bug, and works just fine. Not sure what I did wrong. Possibly didn't order the EEPROM resets correctly.

For anyone else that finds this:

1. Update firmware reversing `INVERT_E0_DIR`
1. Flash it (Helps to change the machine name to be sure)
1. `M502`
1. `M500`
1. your extruder should be reversed.

closed 2020-11-15 by dkowis

ellensp, 2020-11-16:
Did you actually re-flash?
With platform io this compiles and uploads the new firmware to the sd card. The re flash doesn't actually happen until you reboot the controller.  This catches a lot of people out.

dkowis, 2020-11-16:
I am 100% sure I did reflash. Changing the machine name guarantees that the firmware loaded.