MarlinFirmware/Marlin #20154: [BUG] Change Filament change feature uses heater and sensor for E2 when option E1 is selected

zhro, 2020-11-16:
### Configuration Files

Test on latest bugfix (downloaded today):

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5543394/Marlin.zip)

Only the most minimal changes needed to get Marlin to run on my printer with LCD and buttons. Features enabled are setup for dual extruder and sensors.

### Steps to Reproduce

Configuration_adv.h

#define WATCH_TEMP_PERIOD 1

Unplug E2 heater (E2 temp sensor connected must stay connected or an alarm will sound when the printer boots).

Sanity check sensors:

```
LCD:
  Temperature > Nozzle 1: > Preheat End 1

Alarm sounds: "Heating Failed: E1"

Reset.

LCD:
  Temperature > Nozzle 2: > Preheat End 2

Alarm sounds: "Heating Failed: E2"

Reset.
```

ACTUAL TEST:

```
LCD:
  Change Filament > Change Filament E2 > Preheat

Alarm sounds: "Heating Failed: E2"

Reset.

LCD:
  Change Filament > Change Filament E1 > Preheat <--- Heating for E1!

Alarm sounds: "Heating Failed: E2" <--- This is the bug! Uses E2 heater/sensor!

Reset.
```

**Expected behavior:**

Filament options for working with E1 affect E1.

**Actual behavior:**

Filament options for working with E1 affect E2!

Note that filament options for working with E2 correctly affect E2 (this is not a problem with configuration, pins, or swapped cables). All options for pre-heating as well as interacting with E1 and E2 affect the correct tool when connected with serial USB with Repetier.

sjasonsmith, 2020-11-16:
Is this only a problem when pre-heating through the menu? Does it behave correctly on your printer when you set temperatures using M104 over serial?

zhro, 2020-11-16:
The firmware is responding to all of the right commands on the correct tools and reporting from the correct sensors between E1 and E2 when checked through Repetier over serial USB.

This is ONLY a problem when I choose E1 from the CHANGE FILAMENT menu!

sjasonsmith, 2020-11-16:
@zhro I assume you accidentally closed this?

zhro, 2020-11-16:
Yes.

zhro, 2020-11-16:
Re-opened.

sjasonsmith, 2020-11-16:
I wish GitHub would move those buttons further apart! It is especially easy to do from phones.

zhro, 2020-11-16:
I may have been unclear in my report. The "Heating Failed" is not an error but a sanity check to confirm that the preheat commands are going to the correct tool E1 or E2. The alarm will always sound because I set WATCH_TEMP_PERIOD to 1 second.

This is a reference to demonstrate that E1 and E2 is correctly configured when presented with the failure when choosing options from the Change Filament menu.

I am porting Marlin to my printer and I checked, double-checked, and triple-checked that all connections were going to the right pins before I finally realized that this was a bug in the software.

sjasonsmith, 2020-11-16:
I am trying to reproduce this in a simulated environment, and everything seems to work properly.
One detail of your configuration I cannot simulate is the MAX6675 temperature sensor.

Can you try configuring both of your temperature sensors to `998`, and try to reproduce the issue? That is a dummy temperature sensor that always returns 25 degrees, so it should trigger your heating failures.

If the correct hotend is displayed in your failures when using the dummy sensors, then perhaps something unique to your MAX6675 configuration is causing problems.

sjasonsmith, 2020-11-16:
Another difference is that I have `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER` display. I believe these use all the same menus and that should not matter, but maybe the screen ends up changing menus...

zhro, 2020-11-16:
I get a compile error when attempting to use dummy sensor `998`:

```
In file included from Marlin\src\HAL\AVR\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\AVR\HAL.cpp:24:
Marlin\src\HAL\AVR\../../inc/SanityCheck.h:1642:4: error: #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
   #error "TEMP_0_PIN (required for TEMP_SENSOR_0) not defined for this board."
    ^
```

zhro, 2020-11-16:
@sjasonsmith I'm using a basic 20x4 display.

zhro, 2020-11-16:
I merged your changes and Marlin now compiles. I'll report back shortly.

zhro, 2020-11-16:
I have the exact same results using dummy temperature option `998`. I can see the temperature set to 25C but Marlin still reports a problem with the E2 temperature sensor when changing filament on E1.

sjasonsmith, 2020-11-16:
Maybe someone else uses `FF_INTERFACEBOARD` or some other character display and can try to test this.

I have a different character display, but can't simulate it and won't be able to test it right away.

ellensp, 2020-11-16:
tested using provided configs changing  BOARD_RAMPS_14_EEB with REPRAP_DISCOUNT_SMART_CONTROLLER  and disabling DIGIPOT_MCP4018 .
This does say "Heating Failed: E2" when Change Filament > Change Filament E1 > Preheat PLA was selected

ellensp, 2020-11-16:
notes:
In Marlin/src/lcd/menu/menu_filament.cpp

`sprintf_P(cmd, _change_filament_command(), _change_filament_extruder);`
If you examine _change_filament_extruder on a 20x4 it is always 1
but if you do this on a 128x64 it changes from 0 to 1 as expected.  
Cause not found...

ellensp, 2020-11-16:
more notes:
added   SERIAL_ECHOLNPAIR("DEBUG _menu_temp_filament_op: mode=",mode," extruder=",extruder); to top of _menu_temp_filament_op
Results where interesting

On a 128x64 when you enter the Change Filament E1 menu you get
4 lines of DEBUG _menu_temp_filament_op: mode=2 extruder=0
moving up or down that menu on the lcd gives 4 more of this line per row.
If you enter the Change Filament E1 menu you get 
4 lines of DEBUG _menu_temp_filament_op: mode=2 extruder=1
moving up or down that menu on the lcd gives 4 more of this line per row.

But on the 20x4 when you enter the Change Filament E1 menu you get 
DEBUG _menu_temp_filament_op: mode=2 extruder=0.   Correct so far
but this instant you move up or down the menu it changes to one line of 
DEBUG _menu_temp_filament_op: mode=2 extruder=1    This is not correct

ellensp, 2020-11-17:
Well I can't find the cause, need someone with a more experience than I.

ellensp, 2020-11-19:
This has been found and fixed in bugfix, see above PR

idonasch, 2020-12-18:
#20201 fixes a similar issue in the temperature->PreheatXXX menu, but not in the "Change Filament"->Change/Load/Unload submenus. There it is obvious that instead of the selected preheat index the extruder index is used instead. E1 always uses the PREHEAT_1_* settings and E2 always used the PREHEAT_2_* settings. Configuration is a dual extruder with 2 or more PREHEAT_?_* configurations. Tested with bugfix-2.0.x release from 18.Dec.2020. I can supply configuration*.h on request or open e new report?

rhapsodyv, 2020-12-25:
Can you test if this PR fix your issue too? #20154 

Thanks

PR #20201 (2020-11-19): Fix Fil. Change labels with Extruder > 1
### Description

The code from this menu uses `MenuItemBase::itemIndex` to keep track of the current selected extruder. The problem is that it get overwritten `_menu_temp_filament_op` by this loop:
```cpp
    LOOP_L_N(m, PREHEAT_COUNT)
      ACTION_ITEM_N_S(extruder, ui.get_preheat_label(m), MSG_PREHEAT_M, _change_filament_with_preset);
```

If the screen is big enough to hold all the menu, it will not be a problem, because marlin um call `SUBMENU_N_P(s, msg, []{ _menu_temp_filament_op(` using the same parameters, and everything will be reseted to the r [... cut]
PR #20565 (2020-12-28): Fix Change Filament Menu Actions
### Description

Change filament menu double uses MenuItem::index, leading to one menu level overwriting the other. I tried to fix it on #20201, but I broke the preheat material index.

I'm fixing both cases on this PR.

### Benefits

Fix #20553
Fix #20534
Fix #20154
closed 2020-12-25 by rhapsodyv