MarlinFirmware/Marlin #20156: [PINDA 2 WITH SKR1.4] (PROBING WILL NOT COMPLETE)

GioGalo97, 2020-11-16:
https://imgur.com/gallery/2RfGz6x

I am running an skr1.4 with a pinda 2 and it is my first time using ABL. I am able to home but I am not able to run g29. I am getting a Fade height off alarm as soon as I start the printer. When i connect to prontorface I just get:

SENDING:G29
echo:busy: processing
Error:Probing Failed
[ERROR] Error:Probing Failed

GioGalo97, 2020-11-16:
I also get bed leveling off when I start up the printer in the TFT mode.

sjasonsmith, 2020-11-16:
This issue is being closed because it is incomplete. When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. Always attach configuration files, regardless of whether you think they are involved.

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

closed 2020-11-16 by sjasonsmith