MarlinFirmware/Marlin #20163: [BUG] Temperature option to "Preheat All" does not return the user to the "Info" screen like the other choices do

zhro, 2020-11-16:
Test on latest bugfix

Marlin.zip

Most preheat options will return you to the "Info" display after choosing. But the "All" option does not; leading to the believe that nothing happened or that it's broken. The temperature for the extruders and bed does actually turn on but the feedback of returning to the info screen where temperatures are visible, as consistent with the other options, does not occur.

This can result in the user falsely believing that the printer is not heating up when in fact it is. Besides the inconsistent user experience, this failure to provide feedback can be dangerous if the user thinks that the bed and tools are no heating up.

### Steps to Reproduce

These options are available to me from my 20x4 display.

```
Temperature > Preheat PLA > ...

Options:
  Preheat PLA 1 <-- Returns to Info screen
  Preheat PLA End 1 <-- Returns to Info screen
  Preheat PLA 2 <-- Returns to Info screen
  Preheat PLA End 2 <-- Returns to Info screen
  Preheat PLA All <-- (Does nothing/no feedback)
  Preheat PLA End Bed <-- Returns to Info screen
```

**Expected behavior:**

Upon choosing "Preheat PLA All" I would expect to be returned to the "Info" screen as the other options do.

**Actual behavior:**

There is no visual feedback. The screen does not change and there is no indication that the user's choice started heating up the tools or the bed.

GMagician, 2020-11-16:
I tested it now and in my printer it works. I preset "preheat petg" and it return to main menu

zhro, 2020-11-16:
Do you have a 20x4 screen? Please test with the PLA option if you do since that's what I'm reporting as a bug.

thisiskeithb, 2020-11-16:
> I tested it now and in my printer it works. I preset "preheat petg" and it return to main menu
> Not sure I did what you did in your printer....

What's your config? OP's is a dual extruder/hotend setup which enables the "Preheat x All" option so it will preheat both hotends.

thisiskeithb, 2020-11-16:
I doubt this is the correct fix, but you can add `ui.return_to_status();` to [`menu_temperature.cpp`](https://github.com/MarlinFirmware/Marlin/blob/4dd2496530d3d42205aaf874bcb6f356685aa464/Marlin/src/lcd/menu/menu_temperature.cpp#L107-L115) between lines 112 & 113:

```diff
      #elif HAS_MULTI_HOTEND

        HOTEND_LOOP() PREHEAT_ITEMS(editable.int8, e);
        ACTION_ITEM_S(ui.get_preheat_label(m), MSG_PREHEAT_M_ALL, []() {
          TERN_(HAS_HEATED_BED, []{ _preheat_bed(editable.int8); });
          HOTEND_LOOP() thermalManager.setTargetHotend(ui.material_preset[editable.int8].hotend_temp, e);
+         ui.return_to_status();
        });

      #endif
```

Or replace `Marlin\src\lcd\menu\menu_temperature.cpp` with the one attached: [menu_temperature.cpp.zip](https://github.com/MarlinFirmware/Marlin/files/5550018/menu_temperature.cpp.zip)

GMagician, 2020-11-16:
> What's your config?

No single. That's why it works

rhapsodyv, 2020-11-16:
Looking at the code, it will not preheat the bed in that option (`Preheat PLA All`) ! `TERN_(HAS_HEATED_BED, []{ _preheat_bed(editable.int8); });` this just declare a c++ lamba, but don't call it at all. That is the reason it is not returning to status (_preheat_bed will call `ui.return_to_status`).

thisiskeithb, 2020-11-17:
The fix has been merged. Closing.

closed 2020-11-17 by thisiskeithb