MarlinFirmware/Marlin #20168: [BUG] Marlin stops/stuck saying "Media Inserted" or "Printing" randomly

Stefar77, 2020-11-17:
### Bug Description
When printing larger prints the printer randomly stops. (SKR 1.3, tested with all A4988 drivers and TMC5161 drivers)

### Configuration Files
[config.zip](https://github.com/MarlinFirmware/Marlin/files/5551445/config.zip)

If you've made any other modifications describe them in detail here.

### Steps to Reproduce
Print something big.

**Expected behavior:**
Stops moving axis after a random amount of time, hotends stays hot.
TFT still works, connecting to serial still works, I can pause in this state and movement works but it will never unpause.

Expect the firmware to give some sort of error and disable the hotend at least, this is a fire hazard!

Stefar77, 2020-11-17:
** Marlin/src/sd/cardreader.cpp **
void CardReader::release() {
    endFilePrint(); <-- Removing this line fixed my problem (I think)

Stefar77, 2020-11-17:
Currently switched back the TMC5161's and printing some tests.

boelle, 2020-11-17:
Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Stefar77, 2020-11-17:
> Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

I'm running bugfix-2.0.x at the moment with the change mentioned above, without it it fails pretty much every long print.

rhapsodyv, 2020-11-18:
Your problem is related with `SD_DETECT_PIN`. You are using SD from LCD, that uses `EXPA2_04_PIN` (`P1_31`) pin. I would say you have an electrical or cable issue, because the state of the `SD_DETECT_PIN` is changing in the middle of the print.

Try commenting `SD_DETECT_PIN` on `Marlin/src/pins/lpc1768/pins_BTT_SKR_V1_3.h`, line 289.

If it works, try checking everything related with that PIN and with the cable. Try using onboard SD to check if the problem happens too.

Stefar77, 2020-11-18:
> Your problem is related with `SD_DETECT_PIN`. You are using SD from LCD, that uses `EXPA2_04_PIN` (`P1_31`) pin. I would say you have an electrical or cable issue, because the state of the [...]

It's the SD card for sure, just finished a pretty long print without issues.

But it's still an issue that Marlin just does the endFilePrint(); and that doesn't disable the hotend or put an error on the display like 'Interrupted' or something. Also it's still in half printing state because you have a pause and stop option. Pressing the pause makes it say Pause but you cannot un-pause.

You can test this by starting a print and removing the SD card for just a second.

rhapsodyv, 2020-11-18:
Marlin is calling `endFilePrint()` because `SD_DETECT_PIN` is telling Marlin that you don't have any SD inserted, and a cycle after, it is telling the you have the `Media Inserted`. The problem is not `endFilePrint()`, is `SD_DETECT_PIN` issuing wrong values during the print.

And Marlin wasn't supposed to have the SD removed during a print. It will let Marlin with wrong states in memory. I don't think we need to handle that. You can fill an issue about that, but it's really look like a corner case.

sjasonsmith, 2020-11-18:
@rhapsodyv I’m going to reopen this. I agree that if the firmware is aware enough to stop the print on SD removal, it should also probably turn off heaters.

Stefar77, 2020-11-18:
> @rhapsodyv I’m going to reopen this. I agree that if the firmware is aware enough to stop the print on SD removal, it should also probably turn off heaters.

Thanks! I agree. At least if there is an LCD put an error message "Print interrupted" or something.
And it does leave Marlin in a invalid state, if you remark the endFilePrint() maybe it's leaking some filedescriptors but at least it doesn't crash for up to 9 hours if someone happens to get a cheap Chinese display. :-)

Ps. It's not that big of a deal because I'm buying a quality screen later but good error handling is pretty important specially if you are printing something that costs 2 kilo of filament and someone pulls the card for just a second by accident....

rhapsodyv, 2020-11-18:
> @rhapsodyv I’m going to reopen this. I agree that if the firmware is aware enough to stop the print on SD removal, it should also probably turn off heaters.
`endFilePrint` is not "I'm printing, I need end a printing". It is: "I just need to turn off flags of sd printing". It's called always that you removed the SD card, even if you are printing using serial. So we can't just disable the heaters in this method. We can check for sd printing, but we depend on sd detect and it impossible to handle any serial print.

I closed that issue, because that issue is a false alarm. And just said that I don't think we need handle bad hardware. But I also said that if the user want, he can open a issue about that. I just think that this thread already have too much confusion for the issue that we are taking about.

So, I really think it should be closed, and any other issued opened.

rhapsodyv, 2020-11-18:
Our best bet for that is turning on `HOTEND_IDLE_TIMEOUT`. It will stop heaters when hot stay idle for some timeout. It solves the SD card removal, disconnection when printing over serial and any other type of printing interruption.

@Stefar77 Did you try `HOTEND_IDLE_TIMEOUT`?

sjasonsmith, 2020-11-18:
@rhapsodyv pointed out that this is already reported and has been discussed in another issue, I will close this as a duplicate.

sjasonsmith, 2020-11-18:
Duplicate of #15390

closed 2020-11-18 by sjasonsmith

Stefar77, 2020-11-18:
> Our best bet for that is turning on `HOTEND_IDLE_TIMEOUT`. It will stop heaters when hotend stay idle for some timeout. It solves the SD card removal, disconnection when printing over serial and any other type of printing interruption.
> 
> @Stefar77 Did you try `HOTEND_IDLE_TIMEOUT`?

It was indeed still disabled by default and I forgot to enable it and makes this less of a fire hazard. :+1: 
However I don't like the lack of feedback when something stops the printer and the state it enters when it has an SD card issue or 'accident'. 

I started out suspecting the TMC drivers at first because I've seen a lot of people using them having the same issues.
The lack of feedback from the printer is the worst issue, if it says "SD Card error" when it stops it would be so much easier troubleshooting if your printer stops after 2 days of printing and you wasted half a kilo of filament because you think you fixed your driver issue.