MarlinFirmware/Marlin #20169: [BUG] touch rotation not working for classic_ui with MKS Robin nano 1.2 and MKS TFT 24

ShuzzDE, 2020-11-17:
### Bug Description

When compiling Marlin 2.0.7.2 for an MKS TFT 24 connected to a Robin Nano 1.2 Board, you need to #define TFT_ROTATION TFT_ROTATE_180.
However, for classic UI, only the display is rotated, the touch input routines do not take this rotation into account.
While there seems to be code in place in touch_buttons.c, it does not work for some reason.

### Configuration Files

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/5552006/Configuration.zip)

ZIP file contains Marlin config as well as one file I hacked/changed to get it up and running.
This is just an ugly hack, I did not find the real problem in the sources or I'd have put in a pull request. Sorry, not much of a C-guy... ;-)

### Steps to Reproduce

Just add the following lines into your Marlin config and compile for an MKS Robin Nano 1.2 boards with 2.4" MKS TFT display:
#define MKS_ROBIN_TFT24
#define TFT_CLASSIC_UI
#define TFT_ROTATION TFT_ROTATE_180

**Expected behavior:**

Touch buttons are at the correct spot so they correspond with UI buttons.

**Actual behavior:**

Touch buttons are rotated 180° - just like the UI would be if TFT_ROTATION wasn't defined as TFT_ROTATE_180.

#### Additional Information

There is a preprocessor #if directive in touch_buttons.cpp that looks like it should handle this problem, but it doesn't - at least not for me.
I could not figure out what went wrong, but after removing that directive (therefore enabling the code) I was able to get my display's touch sensor running correctly with Marlin.
See the file touch_buttons.cpp in the attached Configuration.zip

github051, 2020-11-17:
Try the latest 2.0.x bugfix. Both the displays are working for me (Classic UI and Color UI).

rhapsodyv, 2020-11-17:
Recently, Touch Calibration Screen for all TFT was merged on bugfix. So, M995 is available in all 3 UI now.

ShuzzDE, 2020-11-17:
I am sorry, I just noticed that I screwed up. I tested against the 2.0.x branch, not bugfix-2.0.X
Obviously did not pay attention while cloning the source.

I apologize for wasting your time.

closed 2020-11-17 by ShuzzDE