MarlinFirmware/Marlin #20171: [BUG] (short description)

dredivan, 2020-11-17:
### Hardware:
Motherbord - MKS SGEN L V1.0 LPC1768
LCD - MKS TS35 V2.0
Latest Marlin 2.0 Bugfix

### Bug Description
Touchscreen don`t work, no sensor buttons controll buttons when use **TFT_CLASSIC_UI** 
like in this giff https://github.com/MarlinFirmware/Marlin/pull/20049
 [image]
I use 

 

https://github.com/MarlinFirmware/Marlin/pull/20049
### Configuration Files
[My configuration.zip](https://drive.google.com/file/d/1o-KKP-7ZufdSBZI3eEEG4sECYu-4709Z/view?usp=sharing)

rhapsodyv, 2020-11-17:
I can't download you configuration, gives this error: "Sorry, the file you have requested does not exist.".

Do you have `TOUCH_SCREEN` and `TOUCH_SCREEN_CALIBRATION` enabled?

dredivan, 2020-11-17:
@rhapsodyv sorry? try again
[config.zip](https://drive.google.com/file/d/163QsFE1xwZHnp7JinYYcTxPEytxZtL5d/view?usp=sharing)

rhapsodyv, 2020-11-17:
To use touch calibration, you need to enable TOUCH_SCREEN and TOUCH_SCREEN_CALIBRATION. You configs don't have it enabled.

closed 2020-11-17 by rhapsodyv