MarlinFirmware/Marlin #20172: [BUG] MKS SGEN L V1.0 LPC1768 MKS TS35 V2.0  - LCD_BED_LEVELING and TFT_COLOR_UI compilation error

dredivan, 2020-11-17:
### Hardware:
Motherbord - MKS SGEN L V1.0 LPC1768
LCD - MKS TS35 V2.0
Latest Marlin 2.0 Bugfix

### Bug Description
I user **MKS TS35 V2.0** and **TFT_COLOR_UI** mode
Firmware don't want compile with **#define LCD_BED_LEVELING** and **#define TFT_COLOR_UI**
If use define #TFT_CLASSIC_UI and #define LCD_BED_LEVELING then all ok

### Configuration Files
[My configuration.zip](https://drive.google.com/file/d/1o-KKP-7ZufdSBZI3eEEG4sECYu-4709Z/view?usp=sharing)

### Log error
Compiling .pio\build\LPC1768\src\src\gcode\temp\M303.cpp.o
Compiling .pio\build\LPC1768\src\src\gcode\units\M82_M83.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\extui\ui_api.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\fontutils.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\lcdprint.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\marlinui.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_advanced.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_bed_corners.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_bed_leveling.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_configuration.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_custom.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_filament.cpp.o
Compiling .pio\build\LPC1768\src\src\lcd\menu\menu_game.cpp.o
In file included from Marlin\src\lcd\menu\../../inc/../core/boards.h:24,
                 from Marlin\src\lcd\menu\../../inc/MarlinConfigPre.h:37,
                 from Marlin\src\lcd\menu\menu_bed_leveling.cpp:27:
Marlin\src\lcd\menu\menu_bed_leveling.cpp: In function 'void _lcd_level_bed_homing_done()':
**Marlin\src\lcd\menu\menu_bed_leveling.cpp:172:32: error: 'touch' was not declared in this scope**
  172 |       TERN_(HAS_GRAPHICAL_TFT, touch.add_control(CLICK, 0, 0, TFT_WIDTH, TFT_HEIGHT));
      |                                ^~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:419:26: note: in definition of macro 'THIRD'
  419 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\lcd\menu\../../inc/../core/macros.h:196:29: note: in expansion of macro '___TERN'
  196 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:195:29: note: in expansion of macro '__TERN'
  195 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:194:29: note: in expansion of macro '_TERN'
  194 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
      |                             ^~~~~
Marlin\src\lcd\menu\menu_bed_leveling.cpp:172:7: note: in expansion of macro 'TERN_'
  172 |       TERN_(HAS_GRAPHICAL_TFT, touch.add_control(CLICK, 0, 0, TFT_WIDTH, TFT_HEIGHT));
      |       ^~~~~
**Marlin\src\lcd\menu\menu_bed_leveling.cpp:172:50: error: 'CLICK' was not declared in this scope**
  172 |       TERN_(HAS_GRAPHICAL_TFT, touch.add_control(CLICK, 0, 0, TFT_WIDTH, TFT_HEIGHT));
      |                                                  ^~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:419:26: note: in definition of macro 'THIRD'
  419 | #define THIRD(a,b,c,...) c
      |                          ^
Marlin\src\lcd\menu\../../inc/../core/macros.h:196:29: note: in expansion of macro '___TERN'
  196 | #define __TERN(T,V...)      ___TERN(_CAT(_NO,T),V)  // Prepend '_NO' to get '_NOT_0' or '_NOT_1'
      |                             ^~~~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:195:29: note: in expansion of macro '__TERN'
  195 | #define _TERN(E,V...)       __TERN(_CAT(T_,E),V)    // Prepend 'T_' to get 'T_0' or 'T_1'
      |                             ^~~~~~
Marlin\src\lcd\menu\../../inc/../core/macros.h:194:29: note: in expansion of macro '_TERN'
  194 | #define TERN_(O,A)          _TERN(_ENA_1(O),,A)     // OPTION converted to A or '<nul>'
      |                             ^~~~~
Marlin\src\lcd\menu\menu_bed_leveling.cpp:172:7: note: in expansion of macro 'TERN_'
  172 |       TERN_(HAS_GRAPHICAL_TFT, touch.add_control(CLICK, 0, 0, TFT_WIDTH, TFT_HEIGHT));
      |       ^~~~~
*** [.pio\build\LPC1768\src\src\lcd\menu\menu_bed_leveling.cpp.o] Error 1
=========================================================================== [FAILED] Took 148.90 seconds ===========================================================================

Environment    Status    Duration
-------------  --------  ------------
LPC1768        FAILED    00:02:28.897

sjasonsmith, 2020-11-17:
I don't see those errors, even when I build using the entire source folder from your zip file.

It does fail, claiming that it is out of RAM space. Switching to TFT_CLASSIC_UI instead of TFT_COLOR_UI made that problem go away.

@rhapsodyv would you expect the color UI to increase RAM usage that much, if it started from here?
```
RAM:   [======    ]  59.2% (used 19368 bytes from 32736 bytes)
Flash: [====      ]  44.0% (used 208996 bytes from 475136 bytes)
```

dredivan, 2020-11-17:
@sjasonsmith I update link in first post
[configuration.zip](https://drive.google.com/file/d/163QsFE1xwZHnp7JinYYcTxPEytxZtL5d/view?usp=sharing)

rhapsodyv, 2020-11-17:
Are you trying to use TFT_COLOR_UI without TOUCH_SCREEN enabled?

rhapsodyv, 2020-11-17:
This will fix the compilation of Color UI without TOUCH_SCREEN: #20178

rhapsodyv, 2020-11-17:
> @rhapsodyv would you expect the color UI to increase RAM usage that much, if it started from here?
> 
> ```
> RAM:   [======    ]  59.2% (used 19368 bytes from 32736 bytes)
> Flash: [====      ]  44.0% (used 208996 bytes from 475136 bytes)
> ```

When I did the S GEN L TFT support, I used the same defaults from S GEN L V2 (that I have one). And I never tried to use it enabled too many features. I will make a PR to lower the RAM usage for that board and UI.

rhapsodyv, 2020-11-17:
I did a PR that will lower the ram usage for this board: #20179 too

rhapsodyv, 2020-11-18:
The fix was merged.

PR #20178 (2020-11-18): Allow COLOR_UI be used without TOUCH_SCREEN
### Description

Fix #20172, allowing Color UI to be used without TOUCH_SCREEN.

### Benefits

Fix #20172
closed 2020-11-18 by rhapsodyv