MarlinFirmware/Marlin #20173: Error Going Home - Marlin 2.0.7.2 - GTT skr 1.4 - TMC2208 UART

Enano79, 2020-11-17:
Hi to all I have an issue wen try to set home to my printer. I have upgrade to skr 1.4 with TMC2208 
Maiboard GTT  skr 1.4
Drivers TMC2208 UART
EndStop Fisical (new and tested before and after install)

Description: Sending home "G28" it stop Randomly, and I have to try Again and Again to set home (normally axises X or Z),
when dont have any error, and all Axises are in home, and start to print. (send the G28 Z0) some times does not touch the EndStop and start to print to high.

and mayor times I get the Error "Print halted" 

this is a test using ProntFace. 

```
SENT: G28
SENT: G92 E0
SENT: M114
SENT: M105
RECV: echo:busy: processing
RECV: echo:busy: processing
SENT: M105
RECV: echo:busy: processing
RECV: echo:busy: processing
RECV: echo:busy: processing
RECV: echo:busy: processing
SENT: M105
RECV: echo:busy: processing
RECV: echo:busy: processing
RECV: Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

```

thisiskeithb, 2020-11-17:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-17 by thisiskeithb

Enano79, 2020-11-17:
Sorry But It get some Timeouts reading I think that is a bug in the reading time

ebraiman, 2020-12-06:
I'm seeing this error as well on SKR 1.4 with BTT TMC2208 V3.0.

Here are my configuration files.

This was downloaded from https://marlinfw.org/meta/download/

Version 2.0.7.2

[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/5649666/Configuration.files.zip)

ebraiman, 2020-12-06:
SENT: G28 X0
SENT: M114
SENT: M105
RECV: echo:busy: processing
RECV: echo:busy: processing
RECV: echo:busy: processing
RECV: Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

ebraiman, 2020-12-07:
This might be related to a tool change in Cura 4.8