MarlinFirmware/Marlin #20174: Y axis 'ignored' in NOZZLE_PROBE_OFFSET - BLTOUCH

Calprofen, 2020-11-17:
### Bug Description

NOZZLE_PROBE_OFFSET for BLTOUCH configuration. Y OFFSET seems to be ignored and positions probe at the same spot on Y axis despite the value is changed dramatically (0 up to 600 values tested, no change). 

### Configuration Files

[Marlin 2.0.7.2 Config Calprofen.zip](https://github.com/MarlinFirmware/Marlin/files/5554212/Marlin.2.0.7.2.Config.Calprofen.zip)

I have followed tutorials online showing how to setup BLTOUCH with Marlin 2.0 no drastic changes have been made. Only one instructed to do so in tutorial.

**Expected behavior:**

Auto home printer on LCD: Printer moves probe to centre of the print bed. Probes bed twice, moves Z axis up and waits for further command.

**Actual behavior:**

Auto home printer on LCD: Printer moves probe to correct X axis position, moves probe to Y axis spot which is the same not matter the value in Config. Finally printer probes bed twice and moves Z axis up and waits for further command.

hamster65, 2020-11-17:
Probe offset can be configured and viewed with M851 and it is stored in eeprom. Always reset your eeprom to defaults (M502) after flashing a new firmware and store it (M500), to transfer settings you changed in the configuration to the eeprom.

Calprofen, 2020-11-17:
Thank you. Unfortunately I am having the same experience after following what you said.

Steps taken after your comment:
Reset EEPROM
Compile Marlin
Upload
Reset EEPROM
Auto Home Printer
Same position for probe is taken for Y axis

I changed the Y value in the config and repeated the same process but same problem occurs.

boelle, 2020-11-17:
@Calprofen did @hamster65 suggestion help?

Please test the [`bugfix-2.0.x branch`](https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x) to see where it stands. If the problem has been resolved then we can close this issue. If the issue isn't resolved yet, then we should investigate further.

Calprofen, 2020-11-17:
Thank you for all the help @hamster65 & @boelle 

Yes resetting the EEPROM has done the trick!

All the best!

closed 2020-11-18 by shitcreek