MarlinFirmware/Marlin #20175: [FR] Onboard/Motherboard-based Thermistor Support

thisiskeithb, 2020-11-17:
### Description

Boards like the Einsy Rambo, SKR E3 Turbo, and motherboard used in the Eryone ER-20 have an onboard 100K thermistor, but motherboard/onboard thermistors are not natively supported on Marlin.

There's a "hacky" way to get readings where you define the onboard thermistor as a `TEMP_SENSOR_CHAMBER` which gives you feedback on the LCD & associated thermal protections, but that's not ideal.

### Feature Workflow

1. Support onboard thermistor with something like `TEMP_SENSOR_MOTHERBOARD`
2. Show temperature over serial and optionally on the LCD
3. Enjoy thermal protections of motherboard-based thermistors!

3DFans, 2020-11-17:
thx @thisiskeithb

ETE-Design, 2020-11-17:
Would also be great if a 2'nd bed could be added as an option... I have 2 Zone's on my bed and right now I use Chamber to show 2'nd Zone Temp... But then it also need a 2'nd Heater Pin defined to make it work...

slowbro, 2021-06-28:
> Would also be great if a 2'nd bed could be added as an option... I have 2 Zone's on my bed and right now I use Chamber to show 2'nd Zone Temp... But then it also need a 2'nd Heater Pin defined to make it work...

The "Redundant Temp Sensor" option I made would probably do this for you.

ETE-Design, 2021-06-28:
@slowbro In this PR? Would also be great with alternative bitmap for it ;-)

thisiskeithb, 2021-06-28:
> Would also be great if a 2'nd bed could be added as an option... I have 2 Zone's on my bed and right now I use Chamber to show 2'nd Zone Temp... But then it also need a 2'nd Heater Pin defined to make it work...

This should be a different Feature Request.

slowbro, 2021-06-28:
> @slowbro In this PR? Would also be great with alternative bitmap for it ;-)

Ah, I misinterpreted your first message, I thought you were just trying to ensure the zones were the same temp. Agreed with @thisiskeithb, you should open a separate feature request for that :)

slowbro, 2021-06-28:
Since I've been recently spending a lot of time in the temperature module, I'll take this and add a new `TEMP_SENSOR_BOARD` option.

thisiskeithb, 2021-07-12:
Closing since the feature request was complete and I'll open a separate ticket if we can't figure out why it's causing lockups.

closed 2021-07-12 by thisiskeithb