MarlinFirmware/Marlin #20180: [BUG] Automatic Bed Leveling failure

roscsa77, 2020-11-17:
### Bug Description

The first point of the ABL (Bilinear) is not correct. It will be closer to the Nozzle by 0.15mm then all others. The first point is wrong at every time. The result of the repeatibility test is 0.007 mm. I've activated the debug leveling feature and collected the result data of 5 ABLs one by one. The maximum difference between the max and min values of the same point was 0.06 mm. If I correct the first point ABL result by G29 W1 I0 J0 Zx+0.15, then everything will be OK.

### Configuration Files

**Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

If you've made any other modifications describe them in detail here.

### Steps to Reproduce

1. Auto Home
2. ABL
3. Auto Home
4. Test Bed Leveling
4.3. Send G1 X20 Y20 Z3 F5000
4.1. Send G1 X20 Y20 Z0 F200
4.2. Check the gap between Nozzle and Bed by A4 paper
4.3. Send G1 X20 Y20 Z3 F500
4.4. Send G1 X215 Y20 Z3 F5000
4.5. Send G1 X215 Y20 Z0 F200
4.6. Check the gap between the Nozzle and the Bed by A4 paper
4.7. Repeat 4.3.-4.6. steps to check the bed leveling at more leveling points

**Expected behavior:**

Correct Bed Leveling

**Actual behavior:**

Wrong Bed Leveling

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [Contributing to Marlin](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md) for additional guidelines.

Configuration:
Creality Ender 3 with Original BLTouch and ultrabase
BTT SKR V1.3 board with TMC2208 drivers

shitcreek, 2020-11-17:
Without providing configuration files, it is difficult to know where to start. For instance, are you using the latest version of Marlin or the latest bugfix?
Please try that out first if you haven't, and check these sources below first before reporting a bug. You may also want to try to search through the issue queue to see if the issue has already been brought up and isn't a duplicate.

For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

boelle, 2020-11-17:
> ### Configuration Files
> **Required:** Include a ZIP file containing `Configuration.h` and `Configuration_adv.h`.

sjasonsmith, 2020-11-18:
There is nothing we can do with this without configuration files. We don't even have actual measurements or mesh values. For now I will close this issue since we are missing required information.

There are some steps you can use to gather additional information over serial, which might help understand leveling problems.

- Download Marlin bugfix-2.0.x to test with the latest code.
- Enable DEBUG_LEVELING_FEATURE and M114_DETAIL and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Send M502 and M500 to ensure your Configurations are applied.
- Issue the command M111 S247 to enable maximum logging.
- Perform a G28 to do your standard homing procedure.
- Do a G29 to probe the bed. This will also enable bed leveling.
- Do some of the moves that revealed problems before. Take notes.
- Copy the log output into a .TXT file and attach it to your next reply.
- Repeat this procedure, if needed, to demonstrate inconsistencies. From these logs we should hopefully get a better idea of what's going on with your machine.

closed 2020-11-18 by sjasonsmith