MarlinFirmware/Marlin #20186: [BUG] SD card lacks error handling

Stefar77, 2020-11-18:
### Bug Description
When there is an issue with the SD card or SD Detect the printer will be put in invalid state.
It stops printing, hotend / spindle stay active and the display says: 'Media Inserted'
The menu still thinks you are printing

My cheap Chinese SD card has some pretty long flat cables and it just isn't as stable with SKR board then the previous Ramps board. 

### Steps to Reproduce
1. Print something
2. Remove SD for 1 second

**Expected behavior:**
Recover printing or at least notice the user there was a problem with the SD somehow.

**Actual behavior:**
It stops printing, keeps the hotend hot and on your project and sits there.
Menu shops option to stop print, pause print, etc.

Only option is to stop print and restart, recovery is impossible even if enabled!

sjasonsmith, 2020-11-18:
Duplicate of #15390

closed 2020-11-18 by sjasonsmith

rhapsodyv, 2020-11-19:
@Stefar77 I just sent a PR that will correctly abort the print, when the media was removed during a printing.