MarlinFirmware/Marlin #20187: Marlin-2.0.7.2-SKR-mini-E3-V2.0 does not work with BTT UPS 24V

atharvbadkas, 2020-11-18:
Wiring:
The UPS is connected to PWR-DET (on the SKR-mini-E3-V2.0 board). 

Configuration:
The Configuration_adv.h is set as in the documentation herein. The POWER_LOSS pin is taken from the source in btt git. It is the PC12 (value 44) configured in Configuration_adv.h (#define POWER_LOSS_PIN PC12).

Reproduce:
When the power outage detection is turned on, the printer immediately after starting a print detects a power failure and doesnt start to print. It is only possible to turn the printer off.

My config. Files

ellensp, 2020-11-18:
PWR-DET has 3 pins, are you sure you are using the left most pin (when power in plug is on left)

atharvbadkas, 2020-11-18:
> PWR-DET has 3 pins, are you sure you are using the left most pin (when power in plug is on left)

Yes, To even make sure UPS works,I checked its voltage levels,When power outage occurs,It goes from LOW to HIGH.
My Hardware is ok but when I start print, In 3-4 Secs power outage occurs and I have to forse shut down the printer.
I think its bug in marlin.

atharvbadkas, 2020-11-18:
I also saw similar issues in bigtreetech GitHub page regarding power outage, So created this issue in marlin, So bug gets fixed from here.

Links of Bigtreetech GitHub page issues -
https://github.com/bigtreetech/BIGTREETECH-MINI-UPS-V2.0/issues/11
https://github.com/bigtreetech/BIGTREETECH-MINI-UPS-V2.0/issues/10

I even tried it for Marlin 2.0.5 version but no luck.

atharvbadkas, 2020-11-19:
To better understand this bug/issue I have made a video.
Link of video and config files -

https://drive.google.com/folderview?id=1XO3e_3qdQjLRWFBe841LIy1TkoSQHiZj

The video starts with printer updating firmware to marlin 2.0.7

bartik, 2020-11-25:
> PWR-DET has 3 pins, are you sure you are using the left most pin (when power in plug is on left)

There I would dare a question regarding this. On the board itself it says PWR-DET (between PS-ON and E0-STOP). But under the board the table says PT-DET between PS-ON and E0-STOP).

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf

atharvbadkas, 2020-11-25:
> There I would dare a question regarding this. On the board itself it says PWR-DET (between PS-ON and E0-STOP). But under the board the table says PT-DET between PS-ON and E0-STOP). [...]

Its just Name incorrectly shown in image, If you trace the PWR-DT PIN in Schematic, It's PC_12 Or Pin value 44, Which is correct...

Jedrusia, 2020-11-27:
I have a identical problem and i spend many time with this problem. My problem exist with TFT35 with numerical LCD emulation mode and when i will print from SD Card plugged for SD slot on mainboard SKR MINI E3 V2.
Now i resolved my problem with printing in graphic mode of TFT and my file with gcode is loading from SD slot from TFT. This is working. If i plug the sd card with my gcode to SD slot on mainboard with graphical mode problem exist too.

atharvbadkas, 2020-11-30:
UPDATE :
Instead of using PC12 for UPS, I tried PA8 for UPS
(Neopixel PIN, Which is also +5V tolerant)

Behaviour When PC12 Used : 
Power Outage Error Occurs as soon as Printer Starts Printing.

Behaviour When PA8 Used : 
Print Starts Normally
But when Power Outage occurs(I Turn OFF Printer mid Printing) It Does not Raise the Nozzle ,
When Power Restores, Printer gives options to resume and cancel print, When Print is resumed, Nozzle go to previous location and stops there forever.

Please HELP!!
Thanks

Jedrusia, 2020-12-02:
I notice this problem to BTT.
I have a answer: 
"Hello 
The config.ini file in the display needs to be modified, you can refer to this content to modify.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/890
Sincerely,
BIQU customer service"

atharvbadkas, 2020-12-03:
> I notice this problem to BTT.
> I have a answer:
> "Hello
> The config.ini file in the display needs to be modified, you can refer to this content to modify.
> [bigtreetech/BIGTREETECH-TouchScreenFirmware#890](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/890)
> Sincerely,
> BIQU customer service"
> 
> Please check that this content is accurate to resolve issue.

I'm using original LCD Display in my Setup, and the issues are related to TFT Displays.
Config.ino file is for TFT Displays, NOT for Original LCD.
I also contacted @bigtreetech , Even when I told them I'm using Original LCD still they want me to edit config.ino file which does not exist in Marlin folder.
.
I have one technical question, STM have logic level 3.3V and UPS output is +5V when detected, Although STM pins are 5V tolerant but do the logic level coincide and do  STM detect the voltage?

Jedrusia, 2020-12-03:
Today i was updated a firmware of my TFT35 V3 E, to newest version from github and i was modify config.ini, but nothing changes ! During each print start (marlin mode or touch mode) from the motherboard's SD reader takes 1-2 sec. printer power supply error (no message is reported in touch mode) and after a few senunds relay 1.2 disconnects the printer power. I turn them on again and in the marlin mode I can continue printing. However, I still can't print anything because another power error is reported ... and so on all the time ... Has anyone tested the SKR MINI E3 V2 board with BTT UPS 1.0 and marlin 2.0.7.2 at all ????
This problem probably does not occur with version 1.2 boards, so I suspect it is a hardware problem (not physical damage to the board but a problem of a different nature). On SD i making a file: PLR.

Jedrusia, 2020-12-07:
If I disconnect signal cable between motherboard SKR MINI E3 V2 and VTT UPS V1 24V, then printing work normally.
What is a problem ?

atharvbadkas, 2020-12-08:
> If I disconnect signal cable between motherboard SKR MINI E3 V2 and VTT UPS V1 24V, then printing work normally.
> What is a problem ?

I tried mailing @bigtreetech  back and forth, In 2-3 weeks, I have 64 mails in conversation with @bigtreetech .
But even they don't know what they are doing and eventually stopped responding to my mails.
In their product page its clearly mentioned that it is supported and still lots of users face problems.
Total Waste of TIME, MONEY and Other resources thanks to @bigtreetech .
I should have gone with Creality's motherboards

Jedrusia, 2020-12-10:
I am no longer responding to my correspondence. It's sad, but as you can see, for BTT the most important thing is money, and not the customer. I decided to change the equipment supplier.

atharvbadkas, 2020-12-11:
> I am no longer responding to my correspondence. It's sad, but as you can see, for BTT the most important thing is money, and not the customer. I decided to change the equipment supplier.

I also tried contacting @bigtreetech , but they eventually stopped responding me about BTT UPS.
Can you let me know which alternative supplier is for @bigtreetech ??

thisiskeithb, 2020-12-11:
> Can you let me know which alternative supplier is for @bigtreetech ??

Please close this issue if you’re no longer interested in troubleshooting the originally reported hardware issue.

atharvbadkas, 2020-12-11:
> > Can you let me know which alternative supplier is for @bigtreetech ??
> 
> Please close this issue if you’re no longer interested in troubleshooting the originally reported hardware issue.

Sure, will do close..just give us some time to find alternative.

thisiskeithb, 2020-12-11:
> Sure, will do close..just give us some time to find alternative.

That’s not what this issue queue is for.

<hr>

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://forum.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2020-12-11 by thisiskeithb