MarlinFirmware/Marlin #20190: BLTouch Crashes in bed when auto-homing

teoDoul, 2020-11-18:
Bug Description

The BLTouch does not stop the Z-axis stepper motor on homing.

My config:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/5559270/Marlin.zip)

Steps to Reproduce

Launch Homing XYZ
Look the z stepper motor to push on the bed
Turn off power
Expected behavior: The z stepper motor should stop

Actual behavior: The z stepper motor does not stop

teoDoul, 2020-11-18:
I know it is answered but i didn't come to any solution from previous ones

ellensp, 2020-11-18:
Please attach pictures of how bltouch is connected to the controller.

teoDoul, 2020-11-18:
The Images

https://imgur.com/a/40xQkrk

ellensp, 2020-11-18:
[image]
Your black/white cable is in backwards

teoDoul, 2020-11-18:
I changed it. Now it still hits the board but only for a seconds. Before it was non-stop. How can I fix that?

ellensp, 2020-11-18:
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/2.0.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- MarlinFW.org [Marlin Documentation](http://marlinfw.org)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to a bug in Marlin, then you should post a [bug report](https://github.com/MarlinFirmware/Marlin/issues/new/choose).

closed 2020-11-18 by ellensp

teoDoul, 2020-11-18:
Ok thanks

xruell, 2020-11-18:
Teaching Tech on youtube has several step by steps for BLTouch .. got mine working great. You will have to use a piece of paper between the glass and nozzle initially to set Z offset. Then trial and error to get the right squish in the initial layer. Once set it really rocks for me.