MarlinFirmware/Marlin #20197: Hotend idle timeout is not reset by new temperature command or change in temperature

nicwilson58, 2020-11-18:
Tested in 2.0.7.2 and `bugfix-2.0.x` branch

no need to attach any configuration as this has be tested in the default configurations versions.  (its not a config thing, this is idle code not using temperature changes in gcode to reset the timeout timer.

###Overview
if the hotend idle timeout is enabled  (#define HOTEND_IDLE_TIMEOUT),  the timeout can occur even though a print has started,  for example, let's say timeout is set to 5*60,  and we are currently at the 4-minute mark and a print is started.  The startup gcode sends new nozzle and bed temperatures,  and then bed leveling begins which takes let's say 2 minutes.  The new temperature commands did not reset the timeout timer,  so one minute into the bed level temperature goes to 0.  The printer is unaware so the print tries to continue but no filament extrudes due to no temperature on the nozzle.

### Steps to Reproduce

1. Enable #define HOTEND_IDLE_TIMEOUT
2. Wait just short of the timeout period
3. send a print
4.  Watch it still timeout, as extruder motor has not started yet but the new temperature commands did not reset the timer.

**Expected behavior:**

new and changed temperature gcodes should reset the idle timeout timer.

**Actual behavior:**

timeout can still occur before extrusion starts.

.

thisiskeithb, 2020-11-18:
Duplicate of https://github.com/MarlinFirmware/Marlin/issues/19610

thisiskeithb, 2020-11-18:
> no need to attach any configuration

No, that's not how reporting an issue works.

When reporting an issue, please be sure to follow all instructions at the top of the template, including debugging steps, and leave the sections headings intact.

For reference, these are the instructions included in the template:

### Before Reporting a Bug

1. Test with the `bugfix-2.0.x` branch to see whether the issue still exists.

2. Get troubleshooting help from the Marlin community to confirm it's a bug and not just a configuration error. Links at https://github.com/MarlinFirmware/Marlin/issues/new/choose

### Instructions

1. Fill out every section of the template below.

2. **Always attach configuration files, regardless of whether you think they are involved.**

3. Read and understand Marlin's Code of Conduct. By filing an Issue, you are expected to comply with it, including treating everyone with respect: https://github.com/MarlinFirmware/Marlin/blob/master/.github/code_of_conduct.md

closed 2020-11-18 by thisiskeithb

nicwilson58, 2020-11-19:
Then forget it,  my steps were clear if you want me to jump through hoops I won’t I was trying to help, so keep your bug I will fix this myself and keep it to myself.